HeyWillow/willow #30: esp tool error while flashing, invalid choice: 'esp32s3'

Xlinx64, 2023-05-16:
I already build the project in docker sucessfully. I use macOS. But when I want to flash it from my host machine I get the following error:
```
Using venv for esptool
esptool.py --chip esp32s3 -p /dev/tty.usbmodem14201 -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 audio.bin
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...
esptool: error: argument --chip/-c: invalid choice: 'esp32s3' (choose from 'auto', 'esp8266', 'esp32')
```
 I used `./utils.sh flash` or `./utils.sh erase-flash` to perform the operation.

kristiankielhofner, 2023-05-16:
Thank you again for your patience with this!

Can you run:

```
source venv/bin/activate
pip freeze
```

And provide the output?

Xlinx64, 2023-05-16:
Here it is:

- bitstring==4.0.2
- cffi==1.15.1
- cryptography==40.0.2
- ecdsa==0.18.0
- esptool==4.5.1
- pycparser==2.21
- pyserial==3.5
- reedsolo==1.6.0
- six==1.16.0

```
usage: esptool [-h] [--chip {auto,esp8266,esp32}] [--port PORT] [--baud BAUD]
               [--before {default_reset,no_reset,no_reset_no_sync}]
               [--after {hard_reset,soft_reset,no_reset}] [--no-stub]
               [--trace] [--override-vddsdio [{1.8V,1.9V,OFF}]]
               {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
               ...

esptool.py v2.8 - ESP8266 ROM Bootloader Utility

positional arguments:
  {load_ram,dump_mem,read_mem,write_mem,write_flash,run,image_info,make_image,elf2image,read_mac,chip_id,flash_id,read_flash_status,write_flash_status,read_flash,verify_flash,erase_flash,erase_region,version}
                        Run esptool {command} -h for additional help
    load_ram            Download an image to RAM and execute
    dump_mem            Dump arbitrary memory to disk
    read_mem            Read arbitrary memory location
    write_mem           Read-modify-write to arbitrary memory location
    write_flash         Write a binary blob to flash
    run                 Run application code in flash
    image_info          Dump headers from an application image
    make_image          Create an application image from binary files
    elf2image           Create an application image from ELF file
    read_mac            Read MAC address from OTP ROM
    chip_id             Read Chip ID from OTP ROM
    flash_id            Read SPI flash manufacturer and device ID
    read_flash_status   Read SPI flash status register
    write_flash_status  Write SPI flash status register
    read_flash          Read SPI flash content
    verify_flash        Verify a binary blob against flash
    erase_flash         Perform Chip Erase on SPI flash
    erase_region        Erase a region of the flash
    version             Print esptool version

options:
  -h, --help            show this help message and exit
  --chip {auto,esp8266,esp32}, -c {auto,esp8266,esp32}
                        Target chip type
  --port PORT, -p PORT  Serial port device
  --baud BAUD, -b BAUD  Serial port baud rate used when flashing/reading
  --before {default_reset,no_reset,no_reset_no_sync}
                        What to do before connecting to the chip
  --after {hard_reset,soft_reset,no_reset}, -a {hard_reset,soft_reset,no_reset}
                        What to do after esptool.py is finished
  --no-stub             Disable launching the flasher stub, only talk to ROM
                        bootloader. Some features will not be available.
  --trace, -t           Enable trace-level output of esptool.py interactions.
  --override-vddsdio [{1.8V,1.9V,OFF}]
                        Override ESP32 VDDSDIO internal voltage regulator (use
                        with care)
```

kristiankielhofner, 2023-05-16:
Ok, there it is! For some reason you end up calling an older version of esptool you must have installed somewhere else on your system (we use 4.5.1 but you also have 2.8 somewhere). I'll look into working around this but can you try tracking down your 2.8 and moving/removing it for now (if that's ok with you)?

kristiankielhofner, 2023-05-16:
Alternatively, you can pull latest main, remove your old venv and run flash again:

```
rm -rf venv
./utils.sh flash
```

kristiankielhofner, 2023-05-16:
Sure!

As a "thank you" for your patience on this, here's a little inside secret tip for you to try with German:

In the Willow Configuration server URI, append this to your URL:

`?model=large&detect_language=True`

`./utils.sh build`

From host:

`./utils.sh flash-app`

Try just speaking to Willow in full German commands/sentences. It will be a little slower but that is the absolutely highest quality model and it will detect your language and any one of >30 other languages. You can also try experimenting with just detect_language=True.

I haven't dealt with German much since my Au Pairs when I was a child but I believe German has some unicode characters that we may not exactly print correctly on the display. We're working on that!

Xlinx64, 2023-05-16:
Awesome, I will try that. 
Yes, there are ä,ü,ö.
Thank you!

kristiankielhofner, 2023-05-16:
Please do, and even though we're way off topic for this issue let me know how it works out!

closed 2023-05-17 by kristiankielhofner