HeyWillow/willow #32: Unable to find image 'willow:latest' locally

mineshaftgap, 2023-05-16:
Following the instructions from https://github.com/toverainc/willow/blob/main/README.md when running `./utils.sh docker ` I get the following error:

```
Unable to find image 'willow:latest' locally
docker: Error response from daemon: pull access denied for willow, repository does not exist or may require 'docker login': denied: requested access to the resource is denied.
See 'docker run --help'.
```

mineshaftgap, 2023-05-16:
I think this might be due to the build not being tagged properly in the `./utils.sh build-docker` step when built using colima instead of Docker Desktop.

mineshaftgap, 2023-05-16:
But `RUN --mount` requires BuildKit so since I don't have the full Docker Desktop install (it has become bloatware on the Mac) it does not work without it. I have commented out the `--mount` lines to see if I can simple build without a cache.

mineshaftgap, 2023-05-16:
Comment out did seem to fix it, but having another build issue that I will open in another issue.

kristiankielhofner, 2023-05-16:
Ok, I'll look out for your new issue and consider what we want to do about this one.

Thanks for working through it!

mineshaftgap, 2023-05-16:
Sure thing, opened new issue here: https://github.com/toverainc/willow/issues/41

mineshaftgap, 2023-05-16:
And thanks for your work!

closed 2023-05-17 by kristiankielhofner

hamishcunningham, 2023-05-17:
> But `RUN --mount` requires BuildKit so since I don't have the full Docker Desktop install (it has become bloatware on the Mac) it does not work without it

I run buildkit without docker desktop (though on ubuntu)