HeyWillow/willow #41: `./utils.sh config` results in Failed to resolve component 'esp-adf-libs'.

mineshaftgap, 2023-05-16:
When trying to build, running the step `./utils.sh config` produces:

...
```
[6/6] lvgl/lvgl (8.3.7)
CMake Error at /opt/esp/idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'esp-adf-libs'.
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/build.cmake:236 (__build_resolve_and_add_req)
  /opt/esp/idf/tools/cmake/build.cmake:237 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/build.cmake:242 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/build.cmake:242 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/willow/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```

mineshaftgap, 2023-05-16:
Here is the log.
[CMakeOutput.log](https://github.com/toverainc/willow/files/11492188/CMakeOutput.log)

kristiankielhofner, 2023-05-16:
Can you try `cd deps/esp-adf && git submodule update --init components/esp-adf-libs` ?

mineshaftgap, 2023-05-16:
This seems to be currently working, but had to run `git config --global --add safe.directory /willow/deps/esp-adf` to make it work.

mineshaftgap, 2023-05-16:
After it the above completes and I run `./utils.sh config` again it produces the same error. I am going to try dumping the running container and trying again with your `git submodule update`

mineshaftgap, 2023-05-16:
Well looks like I keep getting the error, any other ideas?

kristiankielhofner, 2023-05-16:
At this point I have no idea where your environment is so please run `./utils.sh destroy` and start from the top of install in the container.

kristiankielhofner, 2023-05-16:
With that latest commit try re-building your container and try again

mineshaftgap, 2023-05-16:
That did it, `./utils.sh config` completed.

mineshaftgap, 2023-05-16:
Thanks so much!

closed 2023-05-16 by mineshaftgap

kristiankielhofner, 2023-05-16:
That was kind of an obvious one on our part. Glad we could work through it! Let us know if you have any other issues.