HeyWillow/willow #51: watchdog triggered after wake word

hennk, 2023-05-17:
Hi,

first of all, thank you for this very interesting project. I just got an S3-Box, and flashed Willow configured for local detection and Alexa keyword.

It detects the wakeword fine, but after saying my command, it stays on the "Recording command..." screen, until the watchdog fires and reboots the device.

The log up to this point is:

```
I (22:22:43.343) WILLOW: AUDIO_REC_WAKEUP_START

I (22:22:43.512) WILLOW: AUDIO_REC_VAD_START
I (22:22:49.746) WILLOW: AUDIO_REC_VAD_END
I (22:22:49.846) WILLOW: AUDIO_REC_WAKEUP_END
I (22:22:53.629) RECORDER_SR: MN dect quit
E (489720) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (489720) task_wdt:  - IDLE (CPU 1)
E (489720) task_wdt: Tasks currently running:
E (489720) task_wdt: CPU 0: IDLE
E (489720) task_wdt: CPU 1: afe_mase
E (489720) task_wdt: Aborting.

abort() was called at PC 0x420871e0 on core 0

Backtrace: 0x4037a17e:0x3fc9fe50 0x40383c7d:0x3fc9fe70 0x4038bd62:0x3fc9fe90 0x420871e0:0x3fc9ff00 0x4037b1e9:0x3fc9ff30 0x4211b86b:0x3fcb16d0 0x42003ca9:0x3fcb16f0 0x4038506f:0x3fcb1710

```

The device is connected to a Wifi network without access to the internet (only access to the Homeassistant instance).

Any ideas what might be causing this?

Thanks in advance,

Henning

kristiankielhofner, 2023-05-17:
You're welcome, and welcome!

What is your speech timing like between wake word and command? We are investigating some occasional odd timing issues between wake and command with local commands. For an example of timing we know to work, check this demo video from a user earlier today:

https://github.com/toverainc/willow/issues/47

In the meantime, can you send the output of the following commands from the willow directory?

```
git log | head -n1
cat speech_commands/commands_en.txt
```

Note your speech_commands will be your actual speech commands and may have sensitive info but we need to review the exact grammar to make sure there isn't a bug in the python command grammar generation script with your specific entity names that could be triggering this.

stintel, 2023-05-17:
I'm personally hitting this when I give multinet a command that isn't defined. I suspect this is a bug in [esp-sr](https://github.com/espressif/esp-sr), and we should involve them / report it there.

hennk, 2023-05-18:
Thanks for the quick responses!

```
$ git log | head -n1
commit 43d223704516cc3daffc75b9633a3898f10218d1
```
and
```
$ cat speech_commands/commands_en.txt
1 TURN ON TERRAZA DETECT
2 TURN OFF TERRAZA DETECT
3 TURN ON LUCES TERRAZA
4 TURN OFF LUCES TERRAZA
5 TURN ON SHELLYEM NINETYEIGHT CDACEBA
6 TURN OFF SHELLYEM NINETYEIGHT CDACEBA
7 TURN ON TERRAZA RECORDINGS
8 TURN OFF TERRAZA RECORDINGS
9 TURN ON LUCES ENCIMERA IZQUIERDA
10 TURN OFF LUCES ENCIMERA IZQUIERDA
11 TURN ON TIMBRE
12 TURN OFF TIMBRE
13 TURN ON ZERO X ZERO ZERO ZERO DFFFFEE ZERO ZERO
14 TURN OFF ZERO X ZERO ZERO ZERO DFFFFEE ZERO ZERO
15 TURN ON SHELLY ONE EC
16 TURN OFF SHELLY ONE EC
17 TURN ON SHELLY VENTILADOR DORMITORIO
18 TURN OFF SHELLY VENTILADOR DORMITORIO
19 TURN ON AGUA TERRAZA
20 TURN OFF AGUA TERRAZA
21 TURN ON LUZ CASETA
22 TURN OFF LUZ CASETA
23 TURN ON SHELLY ONE DB
24 TURN OFF SHELLY ONE DB
25 TURN ON DETECTOR HUMO NIAS ALARM
26 TURN OFF DETECTOR HUMO NIAS ALARM
27 TURN ON LUCES ENTRADA
28 TURN OFF LUCES ENTRADA
29 TURN ON LUCES COCINA
30 TURN OFF LUCES COCINA
31 TURN ON LUCES ENCIMERA DERECHA
32 TURN OFF LUCES ENCIMERA DERECHA
33 TURN ON ZIGBEE TWO MQTT MAIN JOIN
34 TURN OFF ZIGBEE TWO MQTT MAIN JOIN
35 TURN ON TERRAZA SNAPSHOTS
36 TURN OFF TERRAZA SNAPSHOTS
```

As you can see, my HA entities are in spanish, and I do not expect them to work well (or at all) during SR, for the moment I am just trying everything out.

I will try later if I can manually create a commands_en.txt with english entity names

RJ, 2023-05-18:
I get this too, if I say something that doesn't get recognised from the multinet commands list. Stuck on "Recording command...". I've been rebooting it with the switch on the side.

kristiankielhofner, 2023-05-18:
@hennk - As of now the local speech commands only support English. Although the Spanish text can be (somewhat) generated the actual speech model almost certainly can't match at all on Spanish phonemes. As @stintel and @RJ have reported there is clearly an issue with local command recognition without match. Because you're attempting to use Spanish commands you're almost certainly never going to get a match and thus can trigger this issue very reliably.

I will start throwing gibberish at my local command setup and update everyone soon.

Thanks!

kristiankielhofner, 2023-05-18:
Re-opening. Because this change involves a new patch to a dependency we only do that once during `./utils.sh install`

For the time being (unfortunately) to test this:

From container:

```
cp sdkconfig sdkconfig.mine
git pull
./utils.sh destroy
./utils.sh install
cp sdkconfig.mine sdkconfig
./utils.sh build
```

From host:

`./utils.sh flash`

Note that main also includes #57 to decrease the unrecognized multinet command timeout to 2 seconds. Would appreciate some testing to make sure it's not too short!

stintel, 2023-05-18:
With cfb9b66b3023d696898a58133ea6993e3feead20 I am no longer able to trigger the TWDT. Nice catch!

stintel, 2023-05-18:
With cfb9b66b3023d696898a58133ea6993e3feead20, things seem much better. I can still trigger TWDT timeout, but only when saying a very long sentence. Short, undefined commands no longer seem to trigger it.

hennk, 2023-05-18:
Thanks, I will try again with the latest changes, but probably won't have time until the weekend.

RJ, 2023-05-19:
i did destroy and rebuild - behaving much better now with unrecognised commands in multinet mode 👍

kristiankielhofner, 2023-05-19:
Great news, thanks for checking it out!

Depending on how recent your local repo is we introduced some minor tweaks for multinet as recently as this morning that should make it even better. Feel free to `git pull`, build, and flash if you'd like to try them.

hennk, 2023-05-30:
I can also confirm that the recent changes have improved the behavior, although I am still able to provoke a watchdog trigger in certain situations.

Thanks for the quick responses :)

commit baa66c61 (2023-05-18): The recorder_sr wrapper from ADF for ESP-SR doesn't handle multinet timeout correctly

When multinet can't match on a command it returns timeout

With this change we return AUDIO_REC_COMMAND_DECT to the recorder_sr callback

Then we catch it, display "Unrecognized Command" in red on line 3 and play error tone

Fixes #51