HeyWillow/willow #58: homeassistant conversation API not understanding text when ending with a "."

fridgehead, 2023-05-18:
I've noticed that the public inference server produces sentences with a "." character at the end. It seems that HASS's conversation API isnt able to parse these and just responds with "no_intent_match". 

If you manually Curl the API endpoint with two sentences (for example "turn on the big lights") it wont understand if the sentence ends with a period:

`
curl http://IP:8123/api/conversation/process -H "..." --data-ascii '{"text":"turn off the big lights.","language":"en"}'
-> {"response":{"speech":{"plain":{"speech":"Sorry, I didn't understand that","extra_data":null}},"card":{},"language":"en","response_type":"error","data":{"code":"no_intent_match"}},"conversation_id":null}%

but:
curl http://IP:8123/api/conversation/process -H "..." --data-ascii '{"text":"turn off the big lights","language":"en"}'
is fine

`

HA version:  2023.1.4

kristiankielhofner, 2023-05-18:
This is unfortunate. We don't see this but we're all running 2023.5 and while we already have some HA compatibility code for API support on older versions we haven't yet formalized our official HA version requirements.

We'll have to ponder this one a bit and get back to you.

fridgehead, 2023-05-19:
Not to worry, I've updated my Hass install and it's working properly now (although the assistant still isn't very good!)

kristiankielhofner, 2023-05-19:
Great!

Appreciate that but other users will almost certainly also encounter this so we'll have to come up with a reasonable approach to handle this as we intend to support reasonably up-to-date HA users (starting from 2023.1, most likely).

With time it becomes a non-issue as users will more and more have upgraded to recent HA versions that don't have this issue.

rogerquake, 2023-09-02:
I'm on the latest HA version and I am noticing this also. I created a custom automation to listen for names on songs or albums and then play them. It is always failing though since it adds a period to the end. Any good way to fix this?

nikito, 2023-09-02:
I also have routines to play music by artist or song name and have not run into this. Have you tried triggering your automation from the assistant prompt? Usually HA ignores the periods or punctuations at the end of the sentences.

rogerquake, 2023-09-02:
It was only happening when sending the sentence to play a specific music track on my Logitech Media Server. If I send the value and do a search for a track vs. a specific track title it works fine.

I did however figure out a fix for the specific track issue by doing a replace in the automation:

service: squeezebox.call_method
data:
  command: playlist
  parameters:
    - addalbum
    - ""
    - ""
    - "{{ trigger.slots.album | replace(\".\", \"\") }}"
target:
  entity_id: media_player.office_2
enabled: true

This doesn't work very well since it needs to be so specific. I've settled on doing a search where having the period doesn't matter:
- track.titlesearch= '{{ trigger.slots.album | replace(".", "") }}'

stintel, 2025-04-01:
It appears more recent versions of Home Assistant do no longer have this problem.

closed 2025-04-01 by stintel