HeyWillow/willow #60: Cant Flash - cant find ttyACM0?

robotica72, 2023-05-18:
Everything seems to go well, up to erasing the flash - It cant find the port, but dmesg shows the Espressif and that it is on /dev/ttyACM0.  Looking at /dev - I can see ttyACM0 is there.   If I disconnect the USB cable, ttyACM0 does go away...

Using venv for esptool
esptool.py v4.5.1
Serial port /dev/ttyACM0

A fatal error occurred: Could not open /dev/ttyACM0, the port doesn't exist

Only potential issue - this is a VM, but Im passing the USB through, and I never had issues in the past doing this on any flashing (ESP / etc...)

Any ideas?

kristiankielhofner, 2023-05-18:
Which distro is this?

Can you run:

```
ls -l /dev/ttyACM0
id
```

robotica72, 2023-05-18:
Damn - Was just going to send it and noticed I was running it as my standard user - ran utils.sh in a root shell and its working.

Sorry for the rookie mistake!    and thanks for the quick reply.

robotica72, 2023-05-18:
Different issue on the flash now:

Using venv for esptool
esptool esp32s3 -p /dev/ttyACM0 -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 audio.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e8:22:f8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00191fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00210000 to 0x005f3fff...
Flash will be erased from 0x00710000 to 0x0078ffff...
Compressed 22256 bytes to 13957...
Wrote 22256 bytes (13957 compressed) at 0x00000000 in 0.4 seconds (effective 454.7 kbit/s)...

A fatal error occurred: Packet content transfer stopped (received 24 bytes)

Ideas??

kristiankielhofner, 2023-05-18:
Try creating a .env file in the willow directory with:

`FLASH_BAUD=115200`

and run erase-flash and flash again.

robotica72, 2023-05-18:
No luck ... same issue

Using configuration overrides from .env file
Using venv for esptool
esptool esp32s3 -p /dev/ttyACM0 -b 115200 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 audio.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e8:22:f8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x00191fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00210000 to 0x005f3fff...
Flash will be erased from 0x00710000 to 0x0078ffff...
Compressed 22256 bytes to 13957...
Wrote 22256 bytes (13957 compressed) at 0x00000000 in 0.4 seconds (effective 455.3 kbit/s)...

A fatal error occurred: Packet content transfer stopped (received 24 bytes)

The erase was fine - of course that means the unit has nothing on the LCD now :)  But it just wont flash

kristiankielhofner, 2023-05-19:
We haven't done any testing with USB passthrough or VMs... Here's a workaround for you (for now):

`./utils.sh dist`

Disable USB passthrough to the VM.
Copy `sallow-dist.bin` to your host machine.
Unplug and plug-in your ESP BOX just to be sure.

Then head over to the [ESP web flashing tool](https://espressif.github.io/esptool-js/) (Chrome is best if you have it)

Provide access to the port when prompted
Click the red "Erase flash"
Set flash address to `0x0`
Click chose file, find sallow-dist.bin wherever you put it on your host.
Click Program

stintel, 2023-05-19:
I suspect the USB device disconnects during flash. In another terminal, run `journalctl -f -k` to confirm that. If that's the case, you might be able to work around that by enabling download mode: press the BOOT/CONFIG button (top button on the left side) while powering up the device. Try flashing again after that.

If that doesn't work, you'll have to use the web flashing tool.

robotica72, 2023-05-19:
Strange - I program a lot of ESP chips with the same laptop and cable.  Im having no luck here - Tried the web tool, crashes at 2% - Erased again, enabled download mode, and same thing stops at 1 or 2%

esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: 7c:df:a1:e8:22:f8
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.443s
Compressed 7929856 bytes to 4295190...
Writing at 0x1000... (0%)
Writing at 0x1216f... (0%)
Writing at 0x1d225... (1%)
Writing at 0x287b3... (1%)
Error: Timeout

robotica72, 2023-05-19:
Im going to try a new cable - another attempt got to 4%

Writing at 0x0... (0%)
Writing at 0x1116f... (0%)
Writing at 0x1c225... (1%)
Writing at 0x277b3... (1%)
Writing at 0x2c265... (1%)
Writing at 0x3380e... (2%)
Writing at 0x3f3ea... (2%)
Writing at 0x4ab20... (3%)
Writing at 0x55fd0... (3%)
Writing at 0x5db1e... (3%)
Writing at 0x635f5... (4%)
Error: Timeout

hamishcunningham, 2023-05-19:
I've seen this on S3 chips a bit, and recommend the boot/reset combination if you haven't tried it already

kristiankielhofner, 2023-05-19:
That's really good information! Flashing S3 has always been really reliable to me and I've never had to use the boot/reset trick but I'll add it to the README.

robotica72, 2023-05-19:
Yeah, still no luck - Ive tried the Windows Flasher, web, and of course the utils.sh flasher - the highest a flash has managed to get to is 5% before it fails.  Its usually 2-3% before it fails.

Ive tried 2 machines and countless different cables - and since its been erased, I only have a brick now with a black screen.  Bought from Adafruit and since it was erased and a flash was attempted, Im guessing they wouldnt accept a return on it.

kristiankielhofner, 2023-05-19:
Have you tried the boot/reset button combination as suggested by @stintel and @hamishcunningham :

"I suspect the USB device disconnects during flash. In another terminal, run journalctl -f -k to confirm that. If that's the case, you might be able to work around that by enabling download mode: press the BOOT/CONFIG button (top button on the left side) while powering up the device. Try flashing again after that."

robotica72, 2023-05-19:
I did - Ive tried programming using download mode and standard - Actually, without download mode the device will just loop now between connected and disconnected - If I enter download mode, it at least stays connected.

SamJongenelen, 2023-05-19:
I had the same issue. Only using esptools from CLI worked, all other flashers didn't (multiple web, the Chinese one)

In my case, it is because I didn't erase_flash before first time Willow flash (it's in the manual....).
Your device is indeed boot looping 

The device is recoverable, you can flash the demo bin on 0x0 and then erase_flash.

kristiankielhofner, 2023-05-19:
What's interesting about this issue is I would have expected the build steps would be the most problematic yet here we are.

The ESP BOX is manufactured by Espressif. We don't have anything to do with the hardware other than building software for it. After build, we use bog-standard esptool.py from Esspressif to flash a device that is 100% built and sold by Espressif. One perspective could be we have nothing to do with the flashing failures discussed in this issue.

However, I take ownership of issues users experience even it they're not *technically* our fault. We will work through this one way or another.

kristiankielhofner, 2023-05-19:
@robotica72 "Never had an issue like this, I use ESP32s in a lot of devices I create"

The ESP BOX uses the ESP32 S3 which is more-or-less the latest and greatest from Espressif. I suspect a lot of these issues are due to that. It's an "ESP32" but different enough to also kind of not be one at the same time.

hamishcunningham, 2023-05-19:
A big difference is that the S3 has internal USB support, which means that most dev boards connect USB directly to the chip instead of to a separate UART translator like CP2104. This makes it harder to flash in my experience.

I've seen errors with older esptool versions that went away with later versions but I can't remember the exact numbers :(

kristiankielhofner, 2023-05-19:
Wow, thanks for letting me know! I missed that somehow and yes that's definitely a BIG difference vs ESP32.

By default we're installing the latest esptool.py 4.5.1 in the venv on the host for flashing. It can be overriden in the .env file so might be worth asking users to try different versions for now in case there's a regression or something in the latest-latest.

robotica72, 2023-05-19:
I just had success - I had to use the Python esptools in Windows to flash the firmware and it went through without any issues!  

Maybe it will help others that have issues

kristiankielhofner, 2023-05-19:
Great! Let us know how things go!

None of the devs use Windows personally and I thought we might want to add something to use python esptools natively in Windows for Windows users. I'll just have to dig up something running Windows and work on it...

robotica72, 2023-05-19:
Yeah, I use Linux  a lot but only in VMs - I do 3D work too much and need my Windows tools for Photogrammetry, 3D Scanning, etc...

If I can offer anything to help on the Windows side to save you time, let me know

kristiankielhofner, 2023-05-19:
We'd really appreciate the help with Windows!

From what I remember we'll probably have to have users install Python from the official Windows installer and write a Powershell script? Does that make sense?

kristiankielhofner, 2023-05-19:
No worries! I don't mean to sound preachy but even though it might not look like it on the surface and with demo videos, etc Willow is still VERY young and I want to make sure people understand that.

kristiankielhofner, 2023-05-19:
@robotica72 #70

robotica72, 2023-05-19:
> We'd really appreciate the help with Windows!
> 
> From what I remember we'll probably have to have users install Python from the official Windows installer and write a Powershell script? Does that make sense?

Makes sense - My Python27 is just a plain old install, so could use any deployment I think.  Just needs Python, pip and then pip install for the esptools (and of course the firmware).   The steps are pretty easy, assuming you have the FW file.

Ill joing this conversation from my "normal" GIT account, I use this one for only a few things and didnt mean to use it when I posted.

SamJongenelen, 2023-05-19:
It's young but men is it everything I hoped for...! I do have some issues of course, will report / ask later

ssurovich, 2023-05-19:
Its speedy, no doubt - instant for basic HA intents.   Im already thinking about how I can, finally, move on to having my HA be Jarvis - The voice part is pretty easy now with Willow - Now I need to figure out how to add feedback to my Google Home speakers when Willow executes something.  Ive re-created Jarvis' voice using services out there, but they are all API driven rather than local, so that will add delay to the replies (and a cost ... Im okay with a cost add-on if it works quick).

kristiankielhofner, 2023-05-19:
Thanks! Depending on which rev you have we've improved response times by about 30% in the last 24 hours or so.

We also introduced a new user-configurable VAD timeout (that depending on your rate of speech) is truly "did that really just happen?" fast.

Are you using local or server based speech rec? I only ask because the Willow Inference Server (open source release next week) and current default in the config (our best-effort community hosted server) can transcribe anything and you should be able to do everything you're describing with that.

We have audio output and TTS (WIS already provides TTS) on the roadmap for Willow.

ssurovich, 2023-05-19:
Local for now - Just wanted to get the flash working and test it out at a high level - Hard to believe it an be improved actually!  (Its that good)

I was looking into the Inference Serer and that will be the way to go - I have no isses putting some GTX/RTX card in one of my servers just for that to make sure its quick - Im assuming it will be more about the RAM, so I wont need an RTX 4090 or anything crazy?

kristiankielhofner, 2023-05-19:
We've spent months highly optimizing the Willow Inference Server. Lowest support card is the GTX 1060 3GB to load base, medium, and large simultaneously and with the default model and settings we have on the community hosted server (Whisper medium > 40 languages, beam 1) it can transcribe 3.8 seconds of speech in 588ms on a GTX 1060. That's fairly long for a voice command but gives a general idea. Here are some other benchmarks: 

Device | Model | Beam Size | Speech Duration (ms) | Inference Time (ms) | Realtime Multiple
-- | -- | -- | -- | -- | --
RTX 4090 | large-v2 | 5 | 3840 | 140 | 27x
H100 | large-v2 | 5 | 3840 | 294 | 12x
H100 | large-v2 | 5 | 10688 | 519 | 20x
H100 | large-v2 | 5 | 29248 | 1223 | 23x
GTX 1060 | large-v2 | 5 | 3840 | 1114 | 3x
Tesla P4 | large-v2 | 5 | 3840 | 1099 | 3x
RTX 4090 | medium | 1 | 3840 | 84 | 45x
GTX 1060 | medium | 1 | 3840 | 588 | 6x
Tesla P4 | medium | 1 | 3840 | 586 | 6x
RTX 4090 | medium | 1 | 29248 | 377 | 77x
GTX 1060 | medium | 1 | 29248 | 1612 | 18x
Tesla P4 | medium | 1 | 29248 | 1730 | 16x
RTX 4090 | base | 1 | 180000 | 277 | 648x (not a typo)

I think it's clear we have a different approach to a locally hosted private voice assistant than the rest of the open source ecosystem and buying NVIDIA hardware may be off-putting for some users but I'm going to write up our approach, the thinking behind it, and of course the tangible results.

Long story short the approach up to this point has been to spend ~$120 (at retail prices) per Raspberry Pi as a voice assistant. I don't want to speak ill of projects but the results aren't going to threaten Alexa anytime soon and while we might not have a problem with a jumbled mess of wires, assembly, and managing what is a full Linux computer X number of assistants isn't compelling for most people.

With Willow it's $50 per assistant that can do local recognition. Or if you want Jarvis and are starting from zero you can buy a GPU off eBay (I think GTX 1070 is the sweet spot right now) for $120 and put it in an existing desktop or get something like a Dell Precision off eBay for $200 (I just ordered this exact configuration to demo and document it). So for us, the math breaks down thusly:

Four Raspberry Pis = $480 (MSRP pricing, in reality much more than that)
Something to run Home Assistant = another $120 (let's say Raspberry Pi again)
Total = $600

VS

Four Willows = $200
Machine of your choice from eBay = $200
GTX 1070 = $120
Total = $520

...and meanwhile this machine (with CUDA GPU with spare memory and compute) is going to absolutely crush a Raspberry Pi for Home Assistant and anything else tinkerers like us enjoy experimenting with.

kristiankielhofner, 2023-05-19:
Sorry, too excited to have mentioned this - if we want to move the GPU, etc discussions to GH discussions that would be better as eventually we'll close this issue and it will effectively disappear.

rogerquake, 2023-05-20:
I ran into a failure flashing from my Linux host which I used to build the image using  Docker. So I began to use the method of copying the willow-dist.bin file so I could try and flash on Windows. I was able to erase the device using the ESPWebTool and then flashed it with the bin. However after the device boots up, it's stuck on a screen that says:

Welcome to Willow
Starting up (server)

Any ideas where I could have messed up?

ssurovich, 2023-05-21:
> I ran into a failure flashing from my Linux host which I used to build the image using Docker. So I began to use the method of copying the willow-dist.bin file so I could try and flash on Windows. I [...]

Strange - I didnt have any luck with the Webtool completing, so I used esptools on Windows in Python and it worked great.  Im wondering if that would fix your issue (No idea why, but since it worked well for me... who knows).

rogerquake, 2023-05-21:
Figured it out! The config tool was appending data to items I was configuring such as my Wifi SSID and HomeAssistant URL. When I edited the sdkconfig file manually with pico in the container and re-built, it all worked fine!

MathewJH1, 2023-05-29:
Following all the suggestions I have gotten Willow Flashed across!

But am faced with the following error when I monitor

" (04:20:13.436) WILLOW: [ + ] HTTP client HTTP_STREAM_ON_REQUEST, length=4096
I (04:20:13.439) WILLOW: AUDIO_REC_VAD_END
Total bytes written: 45056
Total bytes written: 47104
I (04:20:13.509) AUDIO_ELEMENT: IN-[http_stream_writer] AEL_IO_DONE,-2
I (04:20:13.510) WILLOW: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (04:20:14.123) HTTP_CLIENT: Body received in fetch header state, 0x3d955d93, 41
I (04:20:14.124) WILLOW: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST
I (04:20:14.128) WILLOW: Got HTTP Response = {"text":"Turn on light.","language":"en"}
I (04:20:14.138) WILLOW: sending command to Home Assistant via WebSocket: {
	"end_stage":	"intent",
	"id":	1685352014,
	"input":	{
		"text":	"Turn on light."
	},
	"start_stage":	"intent",
	"type":	"assist_pipeline/run"
}
I (04:20:14.245) WILLOW: home assistant response_type: error
I (04:20:14.250) WILLOW: received run-end event on WebSocket: {
	"id":	1685352014,
	"type":	"event",
	"event":	{
		"type":	"run-end",
		"data":	null,
		"timestamp":	"2023-05-29T09:20:14.375267+00:00"
	}
}
I (04:20:14.263) AUDIO_THREAD: The play_tone_err task allocate stack on external memory
I (04:20:16.165) AUDIO_ELEMENT: [http_stream_writer] AEL_MSG_CMD_PAUSE
I (04:20:24.271) WILLOW: Wake LCD timeout, turning off LCD
"

sparkydave1981, 2023-05-30:
I tried to flash my first ESPbox last night and had trouble regarding the USB port as well.  I'll have to come back here with the actual error message but it was along the lines of not having access to the port.  I am running in a Terminal window in Ubuntu.  I had to use `sudo` for a lot of commands through the process as I was logged in as a normal? user.  (My Linux experience is limited)

When I executed the `export PORT=/dev/ttyACM0` command, nothing displayed, it just seemed to accept the command and did a line return.  Is that expected?

When I entered the `./utils.sh erase-flash` command I got a PORT error.

kristiankielhofner, 2023-05-30:
Hah yes, this is the issue that will never die ;).

Yes, the export isn't supposed to return anything. General long-standing Unix/Linux convention is to not output anything when a command is successful.

Almost all of these issues come down to permissions. Linux doesn't let regular users write to ports by default so you (usually) either need to be in the ```dialout``` group or use ```sudo``` with the serial port access commands. ```utils.sh``` checks for these things now and will try to (hopefully helpfully) tell you where things are going wrong.

kristiankielhofner, 2023-05-31:
I have updated the help text for these issues. Please run ```git pull``` in your willow directory and read the error messages from these various commands.

Let us know if it is helpful and leads to success or if you have any other additional issues or feedback!

sparkydave1981, 2023-06-02:
Perfect!  That's for the new tips.  I got it flashed.  Now to get it connected to Home Assistant :-)

skrilla0, 2023-06-16:
I was in the process of re-flashing my S3 with the latest updates, and now having the error of finding the USB device. I am also flashing via a VM with USB passthrough, but was fine the first time around, and also just flashed a different S3. just fine.

I've tried  resetting the environment `./utils.sh destroy` and re-building but still no luck. The output from dmesg shows
```
[286570.045746] cdc_acm 2-1:1.0: ttyACM0: USB ACM device
[286573.387236] usb 2-1: USB disconnect, device number 82
[286575.892954] usb 2-1: new full-speed USB device number 83 using xhci_hcd
[286581.480928] usb 2-1: device descriptor read/all, error -110
[286581.816829] usb 2-1: new full-speed USB device number 84 using xhci_hcd
[286582.272759] xhci_hcd 0000:02:1b.0: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
[286587.368799] usb 2-1: device descriptor read/all, error -110
[286587.369550] usb usb2-port1: attempt power cycle
[286588.020713] usb 2-1: new full-speed USB device number 85 using xhci_hcd
[286593.512687] usb 2-1: device descriptor read/8, error -110
[286598.888572] usb 2-1: device descriptor read/8, error -110
```

stintel, 2023-06-16:
I ran into a bunch or USB disconnect issues myself yesterday. Try another cable and/or reduce FLASH_BAUD in utils.sh.

ssurovich, 2023-06-16:
I had nothing but issues with passthrough (and Ive been around VMware since Workstation 1 and ESX 1.7) - Thats why I gave up and went to native Windows to flash, and that work right away.   I flashed about 20 times to test the workflow and it did fail once with a timeout, but it worked right after that.

Have you tried to enter download mode on the ESP before flashing?    That does see to help a lot when you have timeouts.

skrilla0, 2023-06-16:
I'm at a point where it's no longer detecting the USB device when even trying to erase the flash, tried a couple cables, but no luck. dmesg will show the device listed, but ls -l /dev/ttyACM0 won't find it.

hamishcunningham, 2023-06-16:
ls /dev/ttyACM*
too?

skrilla0, 2023-06-16:
> 

Yes, ls /dev/ttyACM* doesn't show the device anymore.

I tried connecting a different S3 that was recently successfully flashed, and host machine is able to identify the USB device fine. When I reconnect the problematic S3, the errors are there: `device descriptor read/8, error -110`

Could I have bricked the device creating this issue?

hamishcunningham, 2023-06-16:
I haven't managed to brick any S3-based boards, but I have seen similar
behaviour where it is *almost* impossible to get them to flash, but then
once flashed they start working again. Mind you I've mostly been working on
boards that have the UF2 bootloader underneath, which is a whole other can
of worms :(

Sorry can't be more helpful!

kimocal, 2023-06-17:
> We haven't done any testing with USB passthrough or VMs... Here's a workaround for you (for now):
> 
> `./utils.sh dist`
> 
> Disable USB passthrough to the VM. Copy `willow-dist.bin` to your host machine. Unplug and plug-in your ESP BOX just to be sure.
> 
> Then head over to the [ESP web flashing tool](https://espressif.github.io/esptool-js/) (Chrome is best if you have it)
> 
> Provide access to the port when prompted Click the red "Erase flash" Set flash address to `0x0` Click chose file, find willow-dist.bin wherever you put it on your host. Click Program

Thank you!  I was having issues with the Box disconnecting in my Ubuntu VM and USB passthrough.   I used the flash_download_tool_3.9.5 from the Espressif website to flash the willow-dist.bin.    

https://www.espressif.com/sites/default/files/tools/flash_download_tool_3.9.5.zip

stintel, 2023-08-17:
Closing as we now offer https://flash.heywillow.io/.

closed 2023-08-17 by stintel