HeyWillow/willow #62: Some helpful 3D prints

mineshaftgap, 2023-05-19:
I have made two very crude 3D prints in order to achieve 2 things:
1. Block out the power LED glow without having to desolder it. For now I have it block the mute LED but not the button, if there is interest I can make another version that doesn't block the mute LED.
2. Stop ESP32-S3-Box from falling with an add on weighted base. I used the approach many have which is to have a place to insert pennies for weight, this right now is geared toward US pennies. If there is interest I can make other diameters if people provide them.

Since Github does not allow uploads of STL files I [zipped them up](https://github.com/toverainc/willow/files/11513051/stl-files.zip)

Here you can see what it looks like. There are a ton of ways it can be improved on, but it already makes it so much more pleasant:
[image]
[image]

kristiankielhofner, 2023-05-19:
Wow, Thanks!

Definitely adding this to the wiki!

For the record I never loved my suggestion of snapping the LEDs off either ;).

mineshaftgap, 2023-05-19:
Excellent, feel free to add it to the repo if that is appropriate. Or I can make a PR if you would like.

mineshaftgap, 2023-05-19:
Also totally understand if you don't want it in the repo since it is not Willow specific.

ssurovich, 2023-05-19:
Great add-on - I was going to look at making something, but always great when someone else does the work before me!  

I think Im going to modify the original case a bit - I dont want the +5v out on the side like it is, going to see if there is any room to add some wires out of the back to make it look "cleaner".

mineshaftgap, 2023-05-19:
@ssurovich I am not a fan of the USB C on the base with the angle it is at, I might look to see if there is a good adapter like one of these to help with the cable management: https://www.amazon.com/AreMe-Extender-Compatible-Notebook-Computers/dp/B0B2NJ3P3L

kristiankielhofner, 2023-05-19:
Nor am I. Eventually we'd like to get some power supplies that make sense - correct angle on the connector, 1 AMP.

For the time being I caved and bought this (totally over-spec and expensive) cable just for the connector:

https://www.amazon.com/dp/B09FK2YXNQ?ref=ppx_yo2ov_dt_b_product_details&th=1

kristiankielhofner, 2023-05-19:
Sorry, too excited to have mentioned this - if we want to move the cable and printing discussions to GH discussions that would be better as eventually we'll close this issue and it will effectively disappear.

mineshaftgap, 2023-05-19:
@kristiankielhofner Absolutely let's move this to discussions, can an issue be migrated to a discussion on Github or should I make a new Discussion there?

mineshaftgap, 2023-05-19:
Oh these look very promising! https://www.amazon.com/AreMe-Adapter-Converter-Connector-MacBook/dp/B0BK4X856X

mineshaftgap, 2023-05-22:
I just checked, looks like an issue can be migrated to Discussions.

blakadder, 2023-05-23:
Just slap these STLs on Printables or Thangs

mineshaftgap, 2023-05-23:
Already have a draft I will make public on Printables soon. That does not help with version control so I will probably just make a repo.

kquinsland, 2023-05-24:
> @kristiankielhofner Absolutely let's move this to discussions, can an issue be migrated to a discussion on Github or should I make a new Discussion there?

They [can be](https://docs.github.com/en/discussions/managing-discussions-for-your-community/moderating-discussions)!

I too am not a fan of the USB-C port on the side of the S3 box modules. I've started thinking about how i'd "fix" that and it should be easy enough with a [dedicated USBC/PD negotiation module](https://www.aliexpress.us/item/3256804376177668.html) and supply power through the pin headers on the bottom of the module.

kristiankielhofner, 2023-05-24:
I don't use the side port on The ESP BOX at all, I'm pretty sure it's only there to support basic functionality when detached from the base. Or am I missing something here?