HeyWillow/willow #68: "One shot" install script

kristiankielhofner, 2023-05-19:
As learned from #60 and elsewhere users can sometimes miss crucial steps provided in the dense documentation.

We should create a "one shot" install script that executes the steps in the proper order where the only remaining "manual" step is the configuration menu.

kristiankielhofner, 2023-05-26:
Dependency - re-architect config menus so users are all of a sudden dropped into the current (fairly overwhelming) configuration menu

sparkydave1981, 2023-05-30:
This would be a great addition to the project.  For the 'config' section, perhaps the script could include asking the user for the wifi credentials and the Home Assistant token / other important requirements so that the user doesn't need to navigate the config menu.

closed 2023-06-16 by kristiankielhofner