HeyWillow/willow #79: E (17:30:07.558) WILLOW: failed to read HTTP POST response

shahraizq, 2023-05-20:
Im getting this error "E (17:30:07.558) WILLOW: failed to read HTTP POST response" after i have flashed it and connected it. any one faced any similar issue and know how to resolve this?

shahraizq, 2023-05-21:
Fixed wrong api key was entered.

kristiankielhofner, 2023-05-21:
We can actually address this by attempting to connect to the API at build and verifying the credentials configured. Leaving open to document and track that.

pthimon, 2023-05-21:
I had the same 401 error when using homeassistant 2023.3.1. I upgraded to 2023.5.3 and restarted willow and I think meant that willow switched to using websockets, and then the requests were authenticated ok. 

(If anyone else has to upgrade honeassistant, make sure you upgrade to 2023.4.6 first, as there seems to be a database migration bug if going straight from 2023.3.1 to 2023.5.3)

stintel, 2023-05-31:
The 401 was totally my bad: 9ceb5d617da79c9b00d7c34b91879c4b882e9c3a.

stintel, 2023-09-13:
With WAS, we're doing a basic connectivity check to the endpoint URL. We will move endpoint support from Willow to WAS in the future, and when we do that we will add authentication to this connectivity check.

closed 2025-02-23 by stintel