HeyWillow/willow #84: Stack Overflow when using multinet

tdcrenshaw, 2023-05-21:
When using multinet, booting willow results in a stack overflow. After disabling multinet, willow works correctly. I tried enabling the willow debug options for more info with no luck. 

'
MC Quantized wakenet9: wakeNet9_v1h24_hiesp_3_0.63_0.635, tigger:v3, mode:2, p:0, (May  5 2023 20:32:52)
I (5752) AFE_SR: wake num: 3, mode: 1, (May  5 2023 20:32:52)

I (6211) wifi:<ba-add>idx:1 (ifx:0, 32:ee:52:44:b0:6a), tid:1, ssn:0, winSize:64
I (23:17:17.205) WILLOW: SNTP client synchronized time to 1684689437
Build fst from commands.                                  

***ERROR*** A stack overflow in task main has been detected.                                                        

Backtrace: 0x4037a192:0x3fcace10 0x40383ff5:0x3fcace30 0x40387a22:0x3fcace50 0x40385d70:0x3fcacec0 0x403840ac:0x3fca
cee0 0x403840a2:0x00000000 |<-CORRUPTED
'

kristiankielhofner, 2023-05-21:
Can you provide your current defined commands and add them to this issue?

```cat speech_commands/commands_en.txt```

tdcrenshaw, 2023-05-21:
absolutely 

'''
1 TURN ON 
2 TURN OFF 
3 TURN ON MIKROTIK FILTER ACCEPT DNS FROM STUDIO VLAN
4 TURN OFF MIKROTIK FILTER ACCEPT DNS FROM STUDIO VLAN
5 TURN ON ADAPTIVE LIGHTING ENTRYWAY LIGHTS
6 TURN OFF ADAPTIVE LIGHTING ENTRYWAY LIGHTS
7 TURN ON KITCHEN ECHO DOT DO NOT DISTURB SWITCH
8 TURN OFF KITCHEN ECHO DOT DO NOT DISTURB SWITCH
9 TURN ON ADAPTIVE LIGHTING ADAPT BRIGHTNESS LIVINGROOM
10 TURN OFF ADAPTIVE LIGHTING ADAPT BRIGHTNESS LIVINGROOM
11 TURN ON ADAPTIVE LIGHTING ADAPT BRIGHTNESS MASTER BEDROOM
12 TURN OFF ADAPTIVE LIGHTING ADAPT BRIGHTNESS MASTER BEDROOM
13 TURN ON CHRISTMAS TREE ON OFF
14 TURN OFF CHRISTMAS TREE ON OFF
15 TURN ON MIKROTIK PORT STUDIO VLAN
16 TURN OFF MIKROTIK PORT STUDIO VLAN
17 TURN ON MIKROTIK NAT TCPANY
18 TURN OFF MIKROTIK NAT TCPANY
19 TURN ON CLOSET SWITCH ETHER FIVE PORT
20 TURN OFF CLOSET SWITCH ETHER FIVE PORT
21 TURN ON OFFICE LIGHT
22 TURN OFF OFFICE LIGHT
23 TURN ON ADAPTIVE LIGHTING ADAPT COLOR OFFICE
24 TURN OFF ADAPTIVE LIGHTING ADAPT COLOR OFFICE
25 TURN ON MASTER SLEEP MODE
26 TURN OFF MASTER SLEEP MODE
27 TURN ON MIKROTIK NAT HOME ASSISTANT
28 TURN OFF MIKROTIK NAT HOME ASSISTANT
29 TURN ON LIVINGROOM ECHO SHOW SHUFFLE SWITCH
30 TURN OFF LIVINGROOM ECHO SHOW SHUFFLE SWITCH
31 TURN ON IRRIGATION RESTART SWITCH
32 TURN OFF IRRIGATION RESTART SWITCH
33 TURN ON SILICON LABS EZSP BEDSIDE LAMPS
34 TURN OFF SILICON LABS EZSP BEDSIDE LAMPS
35 TURN ON ADAPTIVE LIGHTING SLEEP MODE OFFICE
36 TURN OFF ADAPTIVE LIGHTING SLEEP MODE OFFICE
37 TURN ON MIKROTIK IOT WIRELESS PORT
38 TURN OFF MIKROTIK IOT WIRELESS PORT
39 TURN ON TYLERS BEDSIDE LAMP
40 TURN OFF TYLERS BEDSIDE LAMP
41 TURN ON BEDROOM ECHO PLUS DO NOT DISTURB SWITCH
42 TURN OFF BEDROOM ECHO PLUS DO NOT DISTURB SWITCH
43 TURN ON MIKROTIK NAT WIREGUARD TWO
44 TURN OFF MIKROTIK NAT WIREGUARD TWO
45 TURN ON CLOSET SWITCH ETHER EIGHT PORT
46 TURN OFF CLOSET SWITCH ETHER EIGHT PORT
47 TURN ON MIKROTIK PORT WLAN ONE
48 TURN OFF MIKROTIK PORT WLAN ONE
49 TURN ON MIKROTIK ETHER SIX PORT
50 TURN OFF MIKROTIK ETHER SIX PORT
51 TURN ON CLOSET SWITCH SFP ELEVEN PORT
52 TURN OFF CLOSET SWITCH SFP ELEVEN PORT
53 TURN ON MIKROTIK TEST VLAN PORT
54 TURN OFF MIKROTIK TEST VLAN PORT
55 TURN ON MIKROTIK NAT SIP
56 TURN OFF MIKROTIK NAT SIP
57 TURN ON DEEPTHOUGHT
58 TURN OFF DEEPTHOUGHT
59 TURN ON IRRIGATION ZONE ONE
60 TURN OFF IRRIGATION ZONE ONE
61 TURN ON ADAPTIVE LIGHTING SLEEP MODE KITCHEN
62 TURN OFF ADAPTIVE LIGHTING SLEEP MODE KITCHEN
63 TURN ON KITCHEN ECHO DOT REPEAT SWITCH
64 TURN OFF KITCHEN ECHO DOT REPEAT SWITCH
65 TURN ON ADAPTIVE LIGHTING ADAPT BRIGHTNESS GUEST BATHROOM
66 TURN OFF ADAPTIVE LIGHTING ADAPT BRIGHTNESS GUEST BATHROOM
67 TURN ON ADAPTIVE LIGHTING LIVINGROOM
[... 265 lines cut ...]
333 TURN ON FLOOR LAMP LIGHT
334 TURN OFF FLOOR LAMP LIGHT
335 TURN ON ADAPTIVE LIGHTING ADAPT COLOR LIVINGROOM
336 TURN OFF ADAPTIVE LIGHTING ADAPT COLOR LIVINGROOM
337 TURN ON KITCHEN ECHO DOT SHUFFLE SWITCH
338 TURN OFF KITCHEN ECHO DOT SHUFFLE SWITCH
339 TURN ON KITCHEN LAMP LEVEL ON OFF
340 TURN OFF KITCHEN LAMP LEVEL ON OFF
341 TURN ON BEDROOM ECHO PLUS SHUFFLE SWITCH
342 TURN OFF BEDROOM ECHO PLUS SHUFFLE SWITCH
343 TURN ON CLOSET SWITCH PORT SFP ELEVEN
344 TURN OFF CLOSET SWITCH PORT SFP ELEVEN
345 TURN ON THIS DEVICE DO NOT DISTURB SWITCH
346 TURN OFF THIS DEVICE DO NOT DISTURB SWITCH
347 TURN ON TRANSMISSION SWITCH
348 TURN OFF TRANSMISSION SWITCH
349 TURN ON CLOSET SWITCH ETHER SIX PORT
350 TURN OFF CLOSET SWITCH ETHER SIX PORT
351 TURN ON PINEAPPLE LAMP LEVEL LIGHT COLOR ON OFF
352 TURN OFF PINEAPPLE LAMP LEVEL LIGHT COLOR ON OFF
353 TURN ON LIVINGROOM ECHO SHOW DO NOT DISTURB SWITCH
354 TURN OFF LIVINGROOM ECHO SHOW DO NOT DISTURB SWITCH
355 TURN ON PORT
356 TURN OFF PORT
'''

tdcrenshaw, 2023-05-21:
After deleting a some entities, multinet runs correctly. I'm assuming I just had too many commands? Here's a working commands_en.txt, it's now 196 lines long. 

```
1 TURN ON 
2 TURN OFF 
5 TURN ON ADAPTIVE LIGHTING ENTRYWAY LIGHTS
6 TURN OFF ADAPTIVE LIGHTING ENTRYWAY LIGHTS
7 TURN ON KITCHEN ECHO DOT DO NOT DISTURB SWITCH
8 TURN OFF KITCHEN ECHO DOT DO NOT DISTURB SWITCH
9 TURN ON ADAPTIVE LIGHTING ADAPT BRIGHTNESS LIVINGROOM
10 TURN OFF ADAPTIVE LIGHTING ADAPT BRIGHTNESS LIVINGROOM
11 TURN ON ADAPTIVE LIGHTING ADAPT BRIGHTNESS MASTER BEDROOM
12 TURN OFF ADAPTIVE LIGHTING ADAPT BRIGHTNESS MASTER BEDROOM
13 TURN ON CHRISTMAS TREE ON OFF
14 TURN OFF CHRISTMAS TREE ON OFF
21 TURN ON OFFICE LIGHT
22 TURN OFF OFFICE LIGHT
23 TURN ON ADAPTIVE LIGHTING ADAPT COLOR OFFICE
24 TURN OFF ADAPTIVE LIGHTING ADAPT COLOR OFFICE
25 TURN ON MASTER SLEEP MODE
26 TURN OFF MASTER SLEEP MODE
29 TURN ON LIVINGROOM ECHO SHOW SHUFFLE SWITCH
30 TURN OFF LIVINGROOM ECHO SHOW SHUFFLE SWITCH
31 TURN ON IRRIGATION RESTART SWITCH
32 TURN OFF IRRIGATION RESTART SWITCH
33 TURN ON SILICON LABS EZSP BEDSIDE LAMPS
34 TURN OFF SILICON LABS EZSP BEDSIDE LAMPS
35 TURN ON ADAPTIVE LIGHTING SLEEP MODE OFFICE
36 TURN OFF ADAPTIVE LIGHTING SLEEP MODE OFFICE
39 TURN ON TYLERS BEDSIDE LAMP
40 TURN OFF TYLERS BEDSIDE LAMP
41 TURN ON BEDROOM ECHO PLUS DO NOT DISTURB SWITCH
42 TURN OFF BEDROOM ECHO PLUS DO NOT DISTURB SWITCH
57 TURN ON DEEPTHOUGHT
58 TURN OFF DEEPTHOUGHT
59 TURN ON IRRIGATION ZONE ONE
60 TURN OFF IRRIGATION ZONE ONE
61 TURN ON ADAPTIVE LIGHTING SLEEP MODE KITCHEN
62 TURN OFF ADAPTIVE LIGHTING SLEEP MODE KITCHEN
63 TURN ON KITCHEN ECHO DOT REPEAT SWITCH
64 TURN OFF KITCHEN ECHO DOT REPEAT SWITCH
65 TURN ON ADAPTIVE LIGHTING ADAPT BRIGHTNESS GUEST BATHROOM
66 TURN OFF ADAPTIVE LIGHTING ADAPT BRIGHTNESS GUEST BATHROOM
67 TURN ON ADAPTIVE LIGHTING LIVINGROOM
68 TURN OFF ADAPTIVE LIGHTING LIVINGROOM
69 TURN ON KITCHEN REPEAT SWITCH
70 TURN OFF KITCHEN REPEAT SWITCH
75 TURN ON SWITCH STATE
76 TURN OFF SWITCH STATE
77 TURN ON LIVINGROOM TV
78 TURN OFF LIVINGROOM TV
85 TURN ON OUTSIDE LIGHTS
86 TURN OFF OUTSIDE LIGHTS
87 TURN ON PATIO LIGHTS
88 TURN OFF PATIO LIGHTS
89 TURN ON BEDROOM LIGHTS
90 TURN OFF BEDROOM LIGHTS
91 TURN ON SCHEDULE EIGHT HUNDRED AND FORTYTHREE EF
92 TURN OFF SCHEDULE EIGHT HUNDRED AND FORTYTHREE EF
95 TURN ON BEDROOM ECHO PLUS REPEAT SWITCH
96 TURN OFF BEDROOM ECHO PLUS REPEAT SWITCH
97 TURN ON KITCHEN ENTRY LIGHT
98 TURN OFF KITCHEN ENTRY LIGHT
99 TURN ON LIVINGROOM ECHO SHOW REPEAT SWITCH
100 TURN OFF LIVINGROOM ECHO SHOW REPEAT SWITCH
111 TURN ON FRONT PORCH LEFT
112 TURN OFF FRONT PORCH LEFT
115 TURN ON MASTER SUNROOM BULB
116 TURN OFF MASTER SUNROOM BULB
117 TURN ON CHANDELIER
[... 105 lines cut ...]
317 TURN ON ADAPTIVE LIGHTING ADAPT COLOR MASTER BATHROOM
318 TURN OFF ADAPTIVE LIGHTING ADAPT COLOR MASTER BATHROOM
321 TURN ON COUCH LIGHT LEFT LEVEL LIGHT COLOR ON OFF
322 TURN OFF COUCH LIGHT LEFT LEVEL LIGHT COLOR ON OFF
323 TURN ON ADAPTIVE LIGHTING ADAPT BRIGHTNESS KITCHEN
324 TURN OFF ADAPTIVE LIGHTING ADAPT BRIGHTNESS KITCHEN
333 TURN ON FLOOR LAMP LIGHT
334 TURN OFF FLOOR LAMP LIGHT
335 TURN ON ADAPTIVE LIGHTING ADAPT COLOR LIVINGROOM
336 TURN OFF ADAPTIVE LIGHTING ADAPT COLOR LIVINGROOM
337 TURN ON KITCHEN ECHO DOT SHUFFLE SWITCH
338 TURN OFF KITCHEN ECHO DOT SHUFFLE SWITCH
339 TURN ON KITCHEN LAMP LEVEL ON OFF
340 TURN OFF KITCHEN LAMP LEVEL ON OFF
341 TURN ON BEDROOM ECHO PLUS SHUFFLE SWITCH
342 TURN OFF BEDROOM ECHO PLUS SHUFFLE SWITCH
345 TURN ON THIS DEVICE DO NOT DISTURB SWITCH
346 TURN OFF THIS DEVICE DO NOT DISTURB SWITCH
347 TURN ON TRANSMISSION SWITCH
348 TURN OFF TRANSMISSION SWITCH
351 TURN ON PINEAPPLE LAMP LEVEL LIGHT COLOR ON OFF
352 TURN OFF PINEAPPLE LAMP LEVEL LIGHT COLOR ON OFF
353 TURN ON LIVINGROOM ECHO SHOW DO NOT DISTURB SWITCH
354 TURN OFF LIVINGROOM ECHO SHOW DO NOT DISTURB SWITCH
```

stintel, 2023-05-21:
Lovely to see I'm not the only one controlling switch ports via Home Assistant 🤓 356 commands is a lot, I've not managed to test that. I guess we need to increase the main task stack size.

Can you slightly increment CONFIG_MAIN_TASK_STACK_SIZE=16384 in sdkconfig and report back which value is enough?

tdcrenshaw, 2023-05-21:
CONFIG_MAIN_TASK_STACK_SIZE=18084 did the trick as far as preventing a stack overflow with the original command list provided above. However the majority of commands do not work, with the monitor printing 

```
I (02:53:11.594) WILLOW: AUDIO_REC_WAKEUP_START

I (02:53:11.948) WILLOW: AUDIO_REC_VAD_START
TOP 1, command_id: 1, phrase_id: 1, prob: 0.869190
I (02:53:14.267) WILLOW: sending command to Home Assistant via WebSocket: {
        "end_stage":    "intent",
        "id":   1684702394,
        "input":        {
                "text": "Turn on ."
        },
        "start_stage":  "intent",
        "type": "assist_pipeline/run"
}
I (02:53:14.285) WILLOW: Got local command ID: '1'

I (02:53:14.353) WILLOW: home assistant response_type: error
I (02:53:14.359) WILLOW: received run-end event on WebSocket: {
        "id":   1684702394,
        "type": "event",
        "event":        {
                "type": "run-end",
                "data": null,
                "timestamp":    "2023-05-21T20:53:14.416620+00:00"
        }
}
I (02:53:14.385) AUDIO_THREAD: The play_tone_err task allocate stack on external memory
I (02:53:14.452) WILLOW: AUDIO_REC_VAD_END
I (02:53:14.453) WILLOW: AUDIO_REC_WAKEUP_END
```

I had one command recognized and correctly executed, but have been unable to reproduce it. 

What's the best way to go about paring down the command_en.txt? I tried to simple delete the lines I didn't want before flashing, but that didn't seem to work.

kristiankielhofner, 2023-05-22:
We plan on eventually generating the local speech recognition commands on device (for dynamic reconfiguration when entities added/removed/renamed/etc). To do this we will have to change the current approach of generating `main/generated_cmd_multinet.h` which is less than ideal anyway and is certainly at least contributing to this issue if not being the actual root cause.

tcrenshaw-ET, 2023-05-22:
I think we can probably close this issue with the solution being to increment CONFIG_MAIN_TASK_STACK_SIZE if you have a large number of commands. It prevents the stack overflow anyway. 

Enjoying the software so far, keep up the good work!

kristiankielhofner, 2023-05-22:
We're going to test with local instances that have 400 commands and set the necessary main task size to what is required for that.

Generating commands is currently somewhat brittle and as you've noted, very inflexible. One of the many reasons we want to move to on device definition with dynamic updating. For your recognition accuracy make sure that you are always flashing the full flash with ```flash``` and not ```flash-app```. Also, as another sign this is currently less than optimal, if you edit either the speech commands or the generated header directly you will likely completely break accuracy because the command ID in the model speech_commands no longer matches with the index lookup in ```generated_cmd_multinet.h```.

Needless to say local command recognition is very new and it needs some work :).

Thanks!

tdcrenshaw, 2023-05-22:
> Also, as another sign this is currently less than ideal, if you edit either the speech commands or the generated header directly you will likely completely break accuracy because the command ID in the model speech_commands no longer matches with the index lookup in generated_cmd_multinet.h.

Good to know; I tried to turn off my desk lamp and accidentally turned off a SIP firewall rule haha.

kristiankielhofner, 2023-05-22:
Hahahaha that's hilarious. Still getting a lot of sipvicious on 5060 ;)?

closed 2023-06-16 by kristiankielhofner