HeyWillow/willow #87: Multinet and Wakenet slower than Alexa

mineshaftgap, 2023-05-22:
While Wakenet is only trivially slower than Alexa, Multinet is perceptible slower.

I timed a very unscientific video of all three turning on my living room lights and here is what I have found:
- [Alexa](https://github.com/toverainc/willow/assets/957262/67070237-5a9a-4ef1-afc0-a4e2d6f832ec) - 1.28 seconds
- [Wakenet](https://github.com/toverainc/willow/assets/957262/6a18af13-9c4f-48b3-bf35-36abb5d5c644) - 1.58 seconds
- [Multinet](https://github.com/toverainc/willow/assets/957262/ba021d30-fbd9-4b08-800b-d570f27ca2a3) - 3.33 seconds

I have confirmed that the command `TURN ON LIVING ROOM LIGHTS` is in speech_commands/commands_en.txt.

What else do I need to do in order to improve my multinet speeds?

https://github.com/toverainc/willow/assets/957262/67070237-5a9a-4ef1-afc0-a4e2d6f832ec

https://github.com/toverainc/willow/assets/957262/6a18af13-9c4f-48b3-bf35-36abb5d5c644

https://github.com/toverainc/willow/assets/957262/ba021d30-fbd9-4b08-800b-d570f27ca2a3

kristiankielhofner, 2023-05-22:
Generally when I do speed comparisons between Alexa and Willow I set my Willow wake word to Alexa so variations in speech and in this case start of the timer (finger) don't impact results. There's an additional issue in that we've observed the command time with Home Assistant can vary quite a bit between executions (HA has to receive the command from Willow/Alexa and issue to device). What version of HA are you running? If it's 2023.5 we use websockets and they are substantially faster.

In our testing we consistently see the Willow device confirm before Alexa.

Depending on a variety of factors (your total commands, command lengths, etc) multinet is typically slower. We haven't spent as much time optimizing it but we'll get into it at some point.

kristiankielhofner, 2023-05-22:
We suggest people use "Hi ESP" especially if you already have Alexa devices (like I do) for obvious reasons. I only suggested using Alexa with Willow for your Alexa v Willow bake-off :).

The "Hi Lexin" wake word comes from the English pronunciation of the Chinese name of Espressif, from what I understand. The pronunciation is so strange it's almost useless and I myself am only able to trigger it maybe 20% of the time. We may end up just disabling it.

mineshaftgap, 2023-05-22:
Ya, was only trying "HI Lexin" as it was one syllable shorter, will stick to "Hi ESP" and hope for a "Willow" wake word in the future ;).

I will look to change the VAD to try and squeeze out a bit more performance. 

Going to build with "Alexa"/multinet now to see if there is still same delay.

mineshaftgap, 2023-05-22:
Until multinet is optimized it sounds like for right now if people want the quickest response they should use wakenet, if they want to keep things local use multinet.

mineshaftgap, 2023-05-22:
One of the reasons I would love multinet to be faster is I have been trying to make sure all my home automations work when power and even internet go out. So I will have to see if I can live with the multinet lag.

kristiankielhofner, 2023-05-22:
One clarification - you're always using Wakenet. Wakenet is for wake word, multinet is for local speech commands, and server/WIS mode is using a Willow Inference Server.

Speaking of which, we will be releasing WIS tomorrow. Our goal is for this to be all local and as right now, if you are using the demo Tovera hosted WIS, you're just sending your speech to us instead of Amazon. That's not the goal for this project.

For performance with accuracy WIS strongly prefers CUDA. If you have CUDA hardware on your WIS instance a locally hosted WIS is the most private, fastest, and most reliable of all options. WIS has been highly optimized by us for acceleration on Nvidia hardware going back to Pascal. I'll be testing, demoing, and documenting CUDA hardware configurations using $100 used GTX hardware later this week. WIS runs on CPU but the performance isn't any better than other CPU Whisper implementations which is to say it won't be competing with Alexa for speed and accuracy anytime soon.

mineshaftgap, 2023-05-22:
You might have given me reason to dust of my hardly used/old gaming computer...would just want to see how low I could make the energy consumption.

mineshaftgap, 2023-05-22:
I can't remember exactly what the card is but I think it was a GTX 1080? Maybe a 1040?

Oh yes, I was already planning on making it dual boot headless Linux (if I didn't previously) and then putting Proxmox with a HA image. I would be putting it on a Zigbee power outlet that should be able to tell me energy used.

kristiankielhofner, 2023-05-22:
To my knowledge there is no such thing as a GTX 1040 so between those two options I'd say it's a 1080, which should work quite well with WIS.

We've tested WIS in WSL for users who want to use Windows gaming machines but given the option I wouldn't go that route.

mineshaftgap, 2023-05-22:
Heh, I will look at it later and figure out what it is!

mineshaftgap, 2023-05-22:
...you can tell I am not much of a gamer...

mineshaftgap, 2023-05-23:
So it is a NVIDIA GeForce GTX 1070.

kristiankielhofner, 2023-05-23:
Perfect! I'm actually going to be doing benchmarks with my recently ordered GTX 1070 later this week. For new users it's probably the best bang for the buck right now.

mineshaftgap, 2023-05-23:
Have Proxmox installed and ready to WIS at it. Will I be needing to do a VM or will you be releasing a container?

kristiankielhofner, 2023-05-23:
It is containerized but setup is a little more involved than ```docker run``` - much like willow today it is early.

kristiankielhofner, 2023-05-23:
WIS released - [discuss on HN](https://news.ycombinator.com/item?id=36044755)

closed 2023-06-16 by kristiankielhofner