HeyWillow/willow #92: Replace beeps with chimes

kristiankielhofner, 2023-05-23:
We currently beep once for success and twice for failure. Not the worst beep in the world but not the most pleasant either. There are a variety of options online for [low-cost royalty chimes](https://www.pond5.com/search?kw=success-chime&media=sfx&duration_max=5).

Ideally we could do a poll or similar on this issue to determine what the community likes the most for success and error and incorporate them in flash for playback.

mineshaftgap, 2023-05-23:
Why not use some already open source sounds like from https://github.com/coffeeking/linux-a11y-sound-theme

kristiankielhofner, 2023-05-23:
I'm open to anything as long as the licensing and royalty situation is CRYSTAL clear - even for commercial use.

Generally speaking the search interface, quality, and clarity on license terms are more than worth the $20 or whatever we will spend on them.

kristiankielhofner, 2023-05-23:
I considered it but I don't even know where to start!

For spoken error messages on device I've worked with [Allison Smith](https://www.theivrvoice.com/) before but for something like chimes I'd be poking around in the dark.

Finding a firm, contracting, the bewildering array of options, etc sounds like a lot of time... I think something like pond5 is a good middle of the road for now.

bert269, 2023-05-24:
....Or will it be possible to make that an 'selectable option' in the configuration menu? - to select the 'success' and 'failure' sound?

Granted - the sound file probably should be a certain format and size - just another thought.

kristiankielhofner, 2023-05-24:
Given that one of the core principles of Willow is user control we will always try to expose as much as possible to user configuration. So yes, you will be able to add/configure your own chimes. We would just like to have the best default chimes out of the "box".

But yes, we will need to do some sanity checking in the configure/build step to ensure that use provided chimes and other audio is in the expected format (or possibly just run everything through ffmpeg to force it).

closed 2023-05-29 by kristiankielhofner