HeyWillow/willow #101: Q: is multiple language currently working?

SamJongenelen, 2023-05-24:
Hi,

After great success with the English version, I've tried to switch to my native language; Dutch. But, I can't get it to work and now I am wondering: Should I be able to use Dutch? I've read in the docs the WIS should be able to do all Whisper languages - but cant find the place to configure it.

 I can see the ESP is receiving Dutch response from HomeAssistant, and my ESP is sending English.

kristiankielhofner, 2023-05-24:
You have to add ```?detect_language=True``` to your Willow Server URI in configuration. You will then see that WIS is returning the transcript in your language with the ISO code for your language in the response.

We don't currently have this documented for two reasons:

- We haven't yet picked and added a font that includes unicode characters, so depending on the language and specific speech you may see some gibberish on the display. However, as said above the transcription will be accurate.

- Home Assistant has (unfortunately) changed their API on this. With the former API they allowed for a language JSON parameter to be defined at the time of request. They have removed this in more recent versions (2023.5) because they have changed their strategy with regard to multiple language support. It is now defined statically in HA. It should still work with Willow but it's disappointing because we can support cases where any speaker of the > 30 languages Whisper supports can walk up to Willow and use their language. That is no longer the case with Home Assistant, unfortunately.

Please try enabling language detection! I'm very curious to see how it works for you. Another inside tip: if language detection or transcription is poor you can also use the large-v2 model. With both your complete URI will look like this:

```https://infer.tovera.io/api/willow?detect_language=True&model=large```

SamJongenelen, 2023-05-25:
I am sorry for being unclear; I am not running my own WIS but using the cloud one (yours).  Thank you for the answer ! :) 
Ideally, running multiple languages is great. In practice I've found a lot of hit and miss, so I think I want it to run Dutch only in the future. 

Currently I am flashing the query params on your cloud URI to test :)

SamJongenelen, 2023-05-25:
update: it works like a charm! Man this is great software

closed 2023-05-25 by SamJongenelen

SamJongenelen, 2023-05-25:
@kristiankielhofner one question: can I force one language too?

kristiankielhofner, 2023-05-26:
The issue with language detection is (basically) the "sound". Of course the further away you are the more diluted the audio becomes and it gets increasingly difficult to distinguish. The fact that our application only involves (for the most part) VERY short speech segments (a few words, basically) complicates this further.

We'll be revamping the multi-language support in WIS and Willow to allow users to configure a supported language to essentially "hardcode" the anticipated language. This should really help with this issue.

Additionally, Meta has [just released](https://about.fb.com/news/2023/05/ai-massively-multilingual-speech-technology/) a new model that represents a major breakthrough in multiple language support. We will be investigating the use of this model.