HeyWillow/willow #104: Unable to create venv

wizbowes, 2023-05-25:
Following the instructions step for step with a fresh Debian 11.5 build.  The following error is received at the step 
`
./utils.sh erase-flash`

.  I don't know enough about virtual environments to resolve it. 

```
Creating venv for esptool
The virtual environment was not created successfully because ensurepip is not
available.  On Debian/Ubuntu systems, you need to install the python3-venv
package using the following command.

    apt-get install python3-venv

You may need to use sudo with that command.  After installing the python3-venv
package, recreate your virtual environment.

Failing command: ['/home/pi/willow/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip'
```

Following the suggestion above to install venv results in the following when trying to then run erase-flash again:
```

Using venv for esptool
./utils.sh: line 65: venv/bin/activate: No such file or directory

```

kristiankielhofner, 2023-05-26:
Run:

```rm -rf venv```

Then erase-flash again

We don't check for the venv module being installed but we should! I'll leave this open to track checking that.

wizbowes, 2023-05-26:
Thanks for the quick response.  This resolved it and I was able to progress.  From my fresh build I also needed to CHMOD /dev/ttyACM0 before the erase and flash would work.  But now I'm up and running thanks very much.

closed 2023-05-26 by wizbowes

kristiankielhofner, 2023-05-26:
We're going to figure out how to document this across Linux distributions but your ```chmod``` fix will only be temporary as devices are managed dynamically. When you reboot or plug/unplug the device the permissions will reset.

The "generally works" solution across distros is to add your user to the ```dialout``` group.

commit 93150c8e (2023-05-26): Fix #104 - check for actual esptool.py as indication of success, otherwise force remove and create venv