HeyWillow/willow #110: openHAB URL should work properly if it has a trailing slash

obones, 2023-05-26:
When configuring for openHAB, the required parameter is the URL of the openHAB instance.
I initially entered it like this: `http://1.2.3.4:8080/`

This lead to calls to openHAB with a double slash like this: `http://1.2.3.4:8080//rest/voice/interpreters`

Sadly, this always give a 302 HTTP response and does not trigger anything in openHAB

As soon as I have modified the URL not to include the trailing slash, it started giving me the 200 OK expected answer.

I understand that this is likely to be an issue in openHAB itself, but it would be nice if willow was to simply ignore the trailing slash if it is present when concatenating the rest of the URL.

kristiankielhofner, 2023-05-26:
Yes, technically speaking this is an issue in openHAB. Per the HTTP standards and general best practice it's perfectly valid to have multiple slashes. We have a burgeoning relationship with members of the openHAB team and we can raise this with them. However, I'm not a fan of requiring users to upgrade something as significant as openHAB or HA for Willow support.

That said, we take ownership of these little things and the least we can do is update the documentation.

obones, 2023-05-26:
Yes, thanks, updating the documentation is fine by me

closed 2023-05-27 by kristiankielhofner