HeyWillow/willow #114: Support ADF 2.6

kristiankielhofner, 2023-05-26:
Espressif is working on releasing ADF 2.6 which includes newer versions of all components such as ESP-IDF and ESP-SR.

We have a branch to track this and are following it closely as we believe it will bring multiple overall improvements and fixes.

bert269, 2023-05-26:
Is there a way to see what version of ADF (and other components) are being downloaded and installed with the repository? A command, or will it all be in the "Build log"?

hamishcunningham, 2023-05-27:
These are set in Dockerfile:
```bash
$ cd willow
$ grep -s ADF_VER *
Dockerfile:ENV ADF_VER="v2.5"
...
```
Note that there's a patch applied though...

stintel, 2023-09-26:
d921cd0d6867279d7dfb6a3a4dc4d836c9a26fe1

closed 2023-09-26 by stintel