HeyWillow/willow #116: Detected overlap at address: 0x710000 for file: audio.bin

faduchesne, 2023-05-26:
When I try to flash my esp32 Box S3 , I ve got an error.

```
faduchesne@MacBook-Pro-de-Fabien willow % ./utils.sh flash
Using venv for esptool
esptool esp32s3 -p /dev/cu.usbmodem2101 -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 audio.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x710000 for file: audio.bin
```

kristiankielhofner, 2023-05-26:
Did you change something in terms of the speech recognition - select multiple wake words, etc? Are you configured to use local command recognition with Multinet? If so, can you send the output of:

```cat speech_commands/commands_en.txt```

faduchesne, 2023-05-27:
I’ve just change the wake word to Alexa . 
but I return to the original Hi E.S.P and it’work. I succeed in flash the firmware . 
just précise I’m French so my English is quite approximative :)

kristiankielhofner, 2023-05-27:
Hmm, flashing with Alexa should work. Did you select multiple wake words? We don't support that yet.

Your English is great!

faduchesne, 2023-05-27:
I appreciate it ;) I'll try to reinstall it with Alexa wake word . I can't Wait for willow work in French through Whisper.

kristiankielhofner, 2023-05-27:
It does!

Add ```?detect_language=true``` to your Willow Inference Server URI and speak any language Whisper supports.

We're currently working through a major update to WIS so this isn't documented yet - it won't be needed in the next version (but it will be backwards compatible).

faduchesne, 2023-05-27:
Oh thx for the tip man ! you're great !

kristiankielhofner, 2023-05-27:
Of course, let us know how it goes!

faduchesne, 2023-05-27:
for sure !

faduchesne, 2023-05-27:
great !! it works fine ! thx for you're wonderful work !

kristiankielhofner, 2023-05-27:
Glad to hear it, you're welcome!

closed 2023-05-27 by kristiankielhofner