HeyWillow/willow #118: Error in configuraiton

skrilla0, 2023-05-28:
When configuring the docker container receiving an error: 

CMake Error at /opt/esp/idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'audio_board'.
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/build.cmake:241 (__build_resolve_and_add_req)
  /opt/esp/idf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:10 (project)

How can this be resolved?

kristiankielhofner, 2023-05-29:
Are you still experiencing this issue?

I just set up a new project directory and I can't reproduce this.

skrilla0, 2023-05-29:
@kristiankielhofner Thanks for reaching out. Yes, still running into the same issue. 

During the build, there is also this error, and not sure if related? 

```
CMake Error at CMakeLists.txt:7 (include):
  include could not find requested file:

    /willow/deps/esp-adf/CMakeLists.txt
```

skrilla0, 2023-05-29:
@kristiankielhofner Please disregard, I was able to figure it out. Thanks for reaching out.

closed 2023-05-29 by skrilla0

XRlabTech, 2023-06-30:
Hello

I am experiencing the same issue after running : idf.py build. @skrilla0  what was the solution?

This was the following output:

CMake Error at /opt/esp/idf/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'audio_board'.
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/build.cmake:241 (__build_resolve_and_add_req)
  /opt/esp/idf/tools/cmake/build.cmake:518 (__build_expand_requirements)
  /opt/esp/idf/tools/cmake/project.cmake:393 (idf_build_process)
  CMakeLists.txt:16 (project)

kristiankielhofner, 2023-06-30:
I don't know about the issue referenced above but do not run idf.py manually. For the build, configure, etc steps use the appropriate `utils.sh` argument. Any unknown arguments you give it are passed to idf.py for you in the event you need to do something (for whatever reason) with idf.py directly.

XRlabTech, 2023-06-30:
Hi  There,

So I get to this part:

tokennodes@Nebula:~/willow$ sudo ./utils.sh docker
Detecting the Python interpreter
Checking "python" ...
Python 3.8.10
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /opt/esp/python_env/idf4.4_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.23.1/bin
  /opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /opt/esp/python_env/idf4.4_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Running sudo ./utils.sh install gives

Cloning into 'esp-adf'...
fatal: Remote branch  not found in upstream origin

I tried to open the ./utils .sh and edit the remote branch but it seemed to cause more issues.