HeyWillow/willow #121: Willow/ Home Assistant Websocket error

MathewJH1, 2023-05-29:
Following all the suggestions I have gotten Willow Flashed across!

But am faced with the following error when I monitor

" (04:20:13.436) WILLOW: [ + ] HTTP client HTTP_STREAM_ON_REQUEST, length=4096
I (04:20:13.439) WILLOW: AUDIO_REC_VAD_END
Total bytes written: 45056
Total bytes written: 47104
I (04:20:13.509) AUDIO_ELEMENT: IN-[http_stream_writer] AEL_IO_DONE,-2
I (04:20:13.510) WILLOW: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (04:20:14.123) HTTP_CLIENT: Body received in fetch header state, 0x3d955d93, 41
I (04:20:14.124) WILLOW: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST
I (04:20:14.128) WILLOW: Got HTTP Response = {"text":"Turn on light.","language":"en"}
I (04:20:14.138) WILLOW: sending command to Home Assistant via WebSocket: {
	"end_stage":	"intent",
	"id":	1685352014,
	"input":	{
		"text":	"Turn on light."
	},
	"start_stage":	"intent",
	"type":	"assist_pipeline/run"
}
I (04:20:14.245) WILLOW: home assistant response_type: error
I (04:20:14.250) WILLOW: received run-end event on WebSocket: {
	"id":	1685352014,
	"type":	"event",
	"event":	{
		"type":	"run-end",
		"data":	null,
		"timestamp":	"2023-05-29T09:20:14.375267+00:00"
	}
}
I (04:20:14.263) AUDIO_THREAD: The play_tone_err task allocate stack on external memory
I (04:20:16.165) AUDIO_ELEMENT: [http_stream_writer] AEL_MSG_CMD_PAUSE
I (04:20:24.271) WILLOW: Wake LCD timeout, turning off LCD
"

_Originally posted by @MathewJH1 in https://github.com/toverainc/willow/issues/60#issuecomment-1566834175_

MathewJH1, 2023-05-29:
found the fault was user error

closed 2023-05-29 by MathewJH1

ssurovich, 2023-05-29:
:) Glad you found the issue Mathew

kubanm3, 2023-10-19:
@MathewJH1 what was the solution? I have the same problem.

[image]

kubanm3, 2023-10-19:
So the problem was home assistant Assist preffered assistant selected with not english language. Changing it to english made willow work. Then changing it to polish made it not work again.