HeyWillow/willow #131: No matching HA Intent

skrilla0, 2023-05-30:
I was able to successfully flash the ESP-32-S3-BOX, & now attempting to configure connecting with HA. Initially, I was getting a login attempt error in HA from an IP address associated with ESP device, and troubleshooted the issue by re-creating a new long lived access token. I'm no longer getting the login attempt error, but getting the 403 response code, and the error in the LCD screen: No matching HA Intent. Logs from the monitor: 

```
I (17:35:49.464) AUDIO_ELEMENT: IN-[http_stream_writer] AEL_IO_DONE,-2
I (17:35:49.464) WILLOW: [ + ] HTTP client HTTP_STREAM_POST_REQUEST, write end chunked marker
I (17:35:49.986) HTTP_CLIENT: Body received in fetch header state, 0x3d95a94d, 57
I (17:35:49.987) WILLOW: [ + ] HTTP client HTTP_STREAM_FINISH_REQUEST
I (17:35:49.992) WILLOW: Got HTTP Response = {"text":"Turn on lights in the kitchen.","language":"en"}
I (17:35:50.002) WILLOW: HASS URL: http://192.168.1.106:8123/api/conversation/process
I (17:35:50.010) WILLOW: sending '{"text":"Turn on lights in the kitchen.","language":"en"}' to Home Assistant API on 'http://192.168.1.106:8123/api/conversation/process'
I (17:35:50.054) WILLOW: HTTP status='403' content_length='14'
```

kristiankielhofner, 2023-05-30:
Which version of Home Assistant are you using? We support pre-2023.5 and it's detecting the conversation API so you must be using a pre-2023.5 version.

However, these older versions of HA need much more manual configuration to match the speech -> text with actual control in HA.

First step is probably to upgrade to 2023.5 and then after that look at the assist pipeline and add aliases, etc for your expected commands.

nikito, 2023-05-30:
When you created the new token, did you update the willow config with it and reflash the device? If you change the token it won't update on the device on it's own, you need to go through the config again and update the value, then reflash it. 🙂 Error 403 is http talk for "forbidden" which usually means the device trying to access the server is unauthorized.

skrilla0, 2023-05-30:
Thanks again for swift response today :) Yes, I did, rebuilt the config, and re-flashed after updating the token. 

I'm running an older version of HA - 2023.03, so upgrading to the latest now to see if it still persists.

skrilla0, 2023-05-31:
To close the loop, the solution did end up being upgrading to latest HA version, then was receiving a 401 error. Re-generated the long live token, and is now working great :) Looking forward to tinkering with this. Thank you again!

closed 2023-05-31 by skrilla0

stintel, 2023-05-31:
The 401 was totally my bad: 9ceb5d617da79c9b00d7c34b91879c4b882e9c3a.