HeyWillow/willow #135: Allow notifications

skorokithakis, 2023-05-31:
This is a feature request that might be out of scope for the project, but it would be nice if I could POST some data to the Willow box and have it show up on the screen. This would be ideal for notifications and things like that, since it's already sitting on my desk.

kristiankielhofner, 2023-05-31:
Hi!

This is WELL within scope and on the roadmap - the ability to do something as simple as alarms and reminders relies on such functionality.

Broadly speaking, the current plan is to use websockets that support events in conjunction with either a Willow Application Server or a platform (Home Assistant, others) component that emulates WAS.

We may also support things such as generic HTTP post, MQTT, etc for such notifications but that has the potential to get very hairy very quickly and we would need to be very careful in how we approach such mechanisms.

skorokithakis, 2023-05-31:
Hmm, why might POST/MQTT get hairy? If you have a way to display notifications, you don't care how they came in, they all go into the same queue, no?

kristiankielhofner, 2023-05-31:
In our drive for performance, high quality audio, etc we are already pushing the fundamental physical capabilities of the ESP S3 the ESP BOX is based on. A $50 ready-made device comes with some compromises and those are rooted in being based on a microcontroller with extremely limited CPU, RAM, flash, etc.

Every single socket/thread/queue/task/etc needs to be carefully accounted for to only not crash the device, but also not interfere/compete with the primary goal of extremely accurate and reliable high quality wake-word and speech recognition.

We also want to ensure a long support lifetime - we have many planned features, users continually suggest more, and there's the unknown of what the road may bring in X years. We do and will continue to operate under tight tolerances and it would be very easy to make Willow unusable by haphazardly dropping in additional functionality.

Again, I'm not saying no, I'm saying it would need to be as carefully considered as everything else we have today has been so that we can continue to provide the high quality user experience we do today.

skorokithakis, 2023-05-31:
Ahh, yes, you mean from a performance perspective. I work with servers so I forget that, with embedded, the stuff Willow does is already basically magic.

kristiankielhofner, 2023-05-31:
Hah, yes. The ESP32 S3 has two 240 MHz cores with 512kb of IRAM and 8MB of additional PSRAM. 16MB of flash.

My machines otherwise have 512GB of RAM - needless to say very different situation!

stintel, 2023-12-20:
@skorokithakis since you're running Willow 0.3.0-rc, did you also update WAS to use the :rc container image tag? This will give you notification support in a new tab named "Applications" in WAS.

skorokithakis, 2023-12-20:
@stintel I did, it works great, thank you.

closed 2023-12-20 by stintel