HeyWillow/willow #137: Newly added sanity checks break ./utils.sh install

plambe, 2023-05-31:
As the title says, installing within the docker container is broken:

```
-- Chip is esp32s3, /willow/deps/esp-adf/components/esp-adf-libs
-- Current board name is CONFIG_ESP32_S3_BOX_BOARD
CMake Error at main/CMakeLists.txt:47 (message):
  Wi-Fi WPA passphrase must be between 8 and 63 ASCII characters                                                  
                                                                                                                  
                                                                                                                  
-- Configuring incomplete, errors occurred!
See also "/willow/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
root@5c1d189cb36f:/willow# ./utils.sh install
```

On top of that there's no vi inside the container, making my life far harder in working around this issue. 

Even if I add a longer wifi pass, it fails for the next sanity check that was added about 1 hour ago.

plambe, 2023-05-31:
I confirm commit 6247bdd0a2380094cf3f280ebe024a2f4b53f4da works

kristiankielhofner, 2023-05-31:
Just committed a fix - re-opened issue so you can ```git pull```, try to run install again, and report back.

Sorry about that!

plambe, 2023-05-31:
I confirm it works now. Thank you for the rapid reaction!

commit 39a85cc4 (2023-05-31): Fix #137
closed 2023-05-31 by plambe