HeyWillow/willow #138: Tab chars in Dockerfile break building container image with Podman

smithfred, 2023-05-31:
```
STEP 3/10: RUN apt-get -qq update && 	apt-get -qq install 	git 	libusb-1.0-0 	nano 	python-is-python3 	python3 	python3-num2words 	python3-pip 	python3-requests 	python3-venv 	sudo 	tio 	&& rm -rf /var/lib/apt/lists/*
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: InvalidAscii', src/vmm/src/builder.rs:599:42
```

Replacing the tabs with spaces fixed this issue.

Btw, trivial: my IDE picked up that there are also a few files with inconsistent line endings:

```
$ find . -type f -execdir file {} \; | grep terminators
./openhab.h: ASCII text, with no line terminators
./rest.h: ASCII text, with no line terminators
./README.md: ASCII text, with very long lines (719), with CRLF line terminators
./audio_tone_uri.h: C source, ASCII text, with CRLF line terminators
./audio_tone_uri.c: ASCII text, with CRLF line terminators
```

stintel, 2023-05-31:
Works just fine for me:

```
stijn@sylvester willow on  main ❯ podman build -t willow-builder .
STEP 1/10: FROM espressif/idf:release-v4.4
STEP 2/10: ARG DEBIAN_FRONTEND="noninteractive"
--> 6f5bf24ee45a
STEP 3/10: RUN apt-get -qq update &&    apt-get -qq install     git     libusb-1.0-0    nano    python-is-python3       python3         python3-num2words       python3-pip     python3-requests        python3-venv    sudo    tio     && rm -rf /var/lib/apt/lists/*
Selecting previously unselected package python3-six.
(Reading database ... 19564 files and directories currently installed.)
Preparing to unpack .../00-python3-six_1.14.0-2_all.deb ...
Unpacking python3-six (1.14.0-2) ...
Selecting previously unselected package sudo.
Preparing to unpack .../01-sudo_1.8.31-1ubuntu1.5_amd64.deb ...
Unpacking sudo (1.8.31-1ubuntu1.5) ...
Selecting previously unselected package nano.
Preparing to unpack .../02-nano_4.8-1ubuntu1_amd64.deb ...
Unpacking nano (4.8-1ubuntu1) ...
Preparing to unpack .../03-git_1%3a2.25.1-1ubuntu3.11_amd64.deb ...
Unpacking git (1:2.25.1-1ubuntu3.11) over (1:2.25.1-1ubuntu3.10) ...
Selecting previously unselected package python-is-python3.
Preparing to unpack .../04-python-is-python3_3.8.2-4_all.deb ...
Unpacking python-is-python3 (3.8.2-4) ...
Selecting previously unselected package python3-certifi.
Preparing to unpack .../05-python3-certifi_2019.11.28-1_all.deb ...
Unpacking python3-certifi (2019.11.28-1) ...
Selecting previously unselected package python3-chardet.
Preparing to unpack .../06-python3-chardet_3.0.4-4build1_all.deb ...
Unpacking python3-chardet (3.0.4-4build1) ...
Selecting previously unselected package python3-idna.
Preparing to unpack .../07-python3-idna_2.8-1_all.deb ...
Unpacking python3-idna (2.8-1) ...
Selecting previously unselected package python3-num2words.
Preparing to unpack .../08-python3-num2words_0.5.6-1_all.deb ...
Unpacking python3-num2words (0.5.6-1) ...
Selecting previously unselected package python3-urllib3.
Preparing to unpack .../09-python3-urllib3_1.25.8-2ubuntu0.2_all.deb ...
Unpacking python3-urllib3 (1.25.8-2ubuntu0.2) ...
Selecting previously unselected package python3-requests.
Preparing to unpack .../10-python3-requests_2.22.0-2ubuntu1_all.deb ...
Unpacking python3-requests (2.22.0-2ubuntu1) ...
Selecting previously unselected package python3.8-venv.
Preparing to unpack .../11-python3.8-venv_3.8.10-0ubuntu1~20.04.7_amd64.deb ...
Unpacking python3.8-venv (3.8.10-0ubuntu1~20.04.7) ...
Selecting previously unselected package python3-venv.
Preparing to unpack .../12-python3-venv_3.8.2-0ubuntu2_amd64.deb ...
Unpacking python3-venv (3.8.2-0ubuntu2) ...
Selecting previously unselected package tio.
Preparing to unpack .../13-tio_1.32-1_amd64.deb ...
Unpacking tio (1.32-1) ...
Setting up tio (1.32-1) ...
Setting up python3-six (1.14.0-2) ...
Setting up python3-chardet (3.0.4-4build1) ...
Setting up python3-certifi (2019.11.28-1) ...
Setting up python3.8-venv (3.8.10-0ubuntu1~20.04.7) ...
Setting up python3-idna (2.8-1) ...
Setting up python3-num2words (0.5.6-1) ...
Setting up sudo (1.8.31-1ubuntu1.5) ...
Setting up git (1:2.25.1-1ubuntu3.11) ...
Setting up python3-urllib3 (1.25.8-2ubuntu0.2) ...
Setting up nano (4.8-1ubuntu1) ...
update-alternatives: using /bin/nano to provide /usr/bin/editor (editor) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/man1/editor.1.gz because associated file /usr/share/man/man1/nano.1.gz (of link group editor) doesn't exist
update-alternatives: using /bin/nano to provide /usr/bin/pico (pico) in auto mode
update-alternatives: warning: skip creation of /usr/share/man/man1/pico.1.gz because associated file /usr/share/man/man1/nano.1.gz (of link group pico) doesn't exist
Setting up python-is-python3 (3.8.2-4) ...
Setting up python3-venv (3.8.2-0ubuntu2) ...
Setting up python3-requests (2.22.0-2ubuntu1) ...
--> 2003804b9ac6
STEP 4/10: RUN useradd --create-home --uid 1000 build
--> 3d8b88c55869
STEP 5/10: COPY --chown=1000 container.gitconfig /home/build/.gitconfig
--> 62bef2a6ab5c
STEP 6/10: COPY container.gitconfig /root/.gitconfig
--> dade975c7ae0
STEP 7/10: ENV PATH="$PATH:/willow/.local/bin"
--> 158157c38375
STEP 8/10: WORKDIR /willow
--> 9308e4f98d5b
STEP 9/10: ENV ADF_VER="v2.5"
--> e3b205369937
STEP 10/10: RUN   cd /opt/esp/idf &&   curl https://raw.githubusercontent.com/espressif/esp-adf/$ADF_VER/idf_patches/idf_v4.4_freertos.patch | patch -p1
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  3480  100  3480    0     0  15000      0 --:--:-- --:--:-- --:--:-- 14935
patching file components/freertos/include/freertos/task.h
patching file components/freertos/tasks.c
Hunk #1 succeeded at 911 (offset -70 lines).
COMMIT willow-builder
--> 582b099c6801
Successfully tagged localhost/willow-builder:latest
582b099c6801e8cf036dc657148d0db9ce2b875f7acffadbf9b06fe3adb0c558
stijn@sylvester willow on  main ➜
```

Please post the exact command how you tried to build, and podman version details etc.

smithfred, 2023-05-31:
I'm using the `krun` runtime, could be that.

Same Podman ver./build command as you.

stintel, 2023-05-31:
To be honest I have never even heard of krun, but that doesn't really matter in this case. If it works with Docker and Podman for me, I'd say this is a krun problem and they need to fix it?

commit 3a1a0929 (2023-06-01): Dockerfile: use 4 space indentation

Fixes #138.
commit 22a471d9 (2023-06-01): Dockerfile: use 4 space indentation

Fixes #138.
commit 9561fa40 (2023-06-01): Dockerfile: use 4 space indentation

Fixes #138.
commit 16125b7c (2023-06-01): Dockerfile: use 4 space indentation

Fixes #138.
closed 2023-06-02 by kristiankielhofner