HeyWillow/willow #147: Tone when the wake word is detected

kristiankielhofner, 2023-06-04:
### Discussed in https://github.com/toverainc/willow/discussions/145

<div type='discussions-op-text'>

<sup>Originally posted by **obones** June  2, 2023</sup>
Right now, when the wake word is properly detected, only the screen turns on.
But if I'm far away, I can't see the screen and thus have no clue if the wake word got caught up properly.
Other vocal assistants play a tune to indicate they are listening for a command, I think it would be nice to have this in Willow.</div>

kovrom, 2023-06-16:
That would be awesome and very helpful. Do you have any idea when you'll be able to add it? Thank you for the awesome project!

kristiankielhofner, 2023-06-16:
I'd like to try to get this in for the 1.0 release (early July) but we have a lot of very high priority tasks for release.

This issue has been labelled "dynamic configuration" because the biggest improvements in 1.0 will be flashing pre-built binaries from a web browser (no more docker and building!) along with the ability to dynamically update configuration for any number of devices, over the air updates, etc. When supported this functionality will be configurable dynamically through our web interface.

We'll do our best but this may not quite make the cut and may end up just after 1.0.

kovrom, 2023-06-16:
Got it. Any way for us to re-enable it in the code somewhere for now, since you said it was there originally?

kristiankielhofner, 2023-06-17:
You have two issues:

1. Get the preferred tone for wake into the right format and audio level (FLAC, 16bit, 16kHz, mono, normalized to the other .flac tone files). Once you have the files you can place them in spiffs/ui and re-build and flash.

2. Patch playback code into Willow. This would be a sync audio playback thread at wakeup start (AUDIO_REC_WAKEUP_START, cb_ar_event in main.c).

Real time operating system microcontroller development in C has very tight tolerances and no room for error. If it could be done fairly quickly and with little risk to overall stability and performance I probably would have just thrown it in for you by now :).

If you get this implemented we would definitely look at a PR!

hamishcunningham, 2023-07-10:
PR #229 does this. Thanks @kristiankielhofner for the code hints!

commit 14822057 (2023-10-10): treewide: add optional wake confirmation

Closes: #147
Closes: #229
commit f978593b (2023-10-10): treewide: add optional wake confirmation

Closes: #147
Closes: #229
commit 8bbb78f6 (2023-10-10): treewide: add optional wake confirmation

Closes: #147
Closes: #229
commit 04013798 (2023-10-10): treewide: add optional wake confirmation

Closes: #147
Closes: #229
closed 2023-10-10 by kristiankielhofner