HeyWillow/willow #148: TTS using WIS TTS does not process numbers

ingrove, 2023-06-05:
First, this is a great addition to Home Assistant!  Thank you!

I've noticed that the TTS engine doesn't seem to process numbers.  I have a few temperature related intent_scripts responding with numbers and the numbers are not processed.  An example from watching the Tio terminal screen is:

 WILLOW: Using WIS TTS URL 'https://infer.tovera.io/api/tts?speaker=CLB&text=it's currently 70.3 inside.'

Although the screen of the ESP32-S3-BOX displays "it's currently 70.3 inside", what is spoken by WIS TTS is "it's currently inside", without the numbers.  I get the same result if I copy the URL in the browser of my PC running Windows.  It skips the numbers.

nikito, 2023-06-05:
This has actually been discussed on the WIS GitHub here: https://github.com/toverainc/willow-inference-server/issues/78

There's a fix in the works, can see the details there 🙂

kristiankielhofner, 2023-06-05:
The fix is actually already deployed in the wisng branch. I just stood up a beta development server you can test against. Change both of your base inference server endpoints to ```https://wisng.tovera.io``` and you should hear numbers with TTS.

ingrove, 2023-06-05:
Thank you for the update.  It seems there is still a problem with numbers that include decimals.  Whole numbers like 68 work fine, however something like 68.9 is silent on the number.

https://wisng.tovera.io/api/tts?speaker=CLB&text=it's currently 68.9 inside.

nikito, 2023-06-05:
Just tested the above and can confirm the same behavior on my local instance.

kristiankielhofner, 2023-06-07:
I added and comment regarding the removal of the currently unused function and I will be further testing and reviewing this today.

closed 2023-06-16 by kristiankielhofner