HeyWillow/willow #150: Latest updates - constant GET /api/websocket HTTP/1.1 errors

bert269, 2023-06-05:
Hi

Updated this morning from the latest Github build and receive these errors - constantly:
E (00:19:37.080) TRANSPORT_WS: Error read response for Upgrade header GET /api/websocket HTTP/1.1
Connection: Upgrade
Host: 192.168.1.233:8123
User-Agent: ESP32 Websocket Client
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: SQB//biJbHbqZUjGQygOhg==

E (00:19:37.095) WEBSOCKET_CLIENT: Error transport connect
I (00:19:37.103) WILLOW: WS event ID: 2
E (00:19:57.107) esp-tls: [sock=54] select() timeout
E (00:19:57.109) TRANSPORT_BASE: Failed to open a new connection: 32774
E (00:19:57.110) TRANSPORT_WS: Error connecting to host 192.168.1.233:8123
E (00:19:57.118) WEBSOCKET_CLIENT: Error transport connect
I (00:19:57.124) WILLOW: WS event ID: 2
E (00:20:17.131) esp-tls: [sock=54] select() timeout
E (00:20:17.133) TRANSPORT_BASE: Failed to open a new connection: 32774
E (00:20:17.134) TRANSPORT_WS: Error connecting to host 192.168.1.233:8123
E (00:20:17.142) WEBSOCKET_CLIENT: Error transport connect
I (00:20:17.148) WILLOW: WS event ID: 2
E (00:20:37.154) esp-tls: [sock=54] select() timeout
E (00:20:37.156) TRANSPORT_BASE: Failed to open a new connection: 32774
E (00:20:37.157) TRANSPORT_WS: Error connecting to host 192.168.1.233:8123
E (00:20:37.165) WEBSOCKET_CLIENT: Error transport connect
I (00:20:37.171) WILLOW: WS event ID: 2
E (00:20:57.179) esp-tls: [sock=54] select() timeout
E (00:20:57.180) TRANSPORT_BASE: Failed to open a new connection: 32774
E (00:20:57.181) TRANSPORT_WS: Error connecting to host 192.168.1.233:8123
E (00:20:57.189) WEBSOCKET_CLIENT: Error transport connect

Device seems to respond ok, so mainly informational ?

kristiankielhofner, 2023-06-05:
Our suspicion is your websocket connection to HA is getting terminated/reset.

Is there a firewall or anything like it between Willow and HA? Can you try a long-running ping to the IP of your Willow device to see if there are any drops?

bert269, 2023-06-05:
Update: by the time I receive this post for a ping request - the ESP-Box has stopped the dropping of connections and continue as normal.
I restarted it and it did not happen again. I will keep an eye on it and as soon as it happens again, quit the monitor and run the ping.

kristiankielhofner, 2023-06-07:
Do you consider this issue resolved at this point?

kristiankielhofner, 2023-06-07:
Fantastic - thanks for working through this with us!

closed 2023-06-07 by kristiankielhofner