HeyWillow/willow #159: TTS Audio Issues

nikito, 2023-06-09:
When getting certain TTS responses, I either see errors on the console, or the output is truncated. Looking at code I see in hass.c the home assistant TTS response seems to be limited to 64 characters (not sure if that is due to a hardware limitation?) which I believe is the reason for the truncated responses, however the other random errors I am uncertain of as the text is well below 64 characters. Here's a snapshot of an example that breaks:
[image]

kristiankielhofner, 2023-06-09:
Can you try building Willow with higher HASS_SPEECH_MAX_LEN defined? It uses `strncpy` but it should be fine with higher values. 

In terms of the additional spaces, that's generally "not good" and I can't think of a reason why someone would want text strings with extra whitespace passed to the model. This would be easy enough to strip on the WIS side.

In terms of caching, we do need to implement better cache control. What we've been doing ourselves is just `rm -rf nginx/cache` but nginx supports the ability to [define routes for cache control](http://nginx.org/en/docs/http/ngx_http_proxy_module.html#proxy_cache_purge) that allow for network requests to purge cache items. However, this is only available with a commercial subscription :(. I've seen some config hacks floating around to kind of attempt this with the open source version of nginx but they're not pretty...

There is also the [ability to set max_length with SpeechT5](https://huggingface.co/docs/transformers/model_doc/speecht5#transformers.SpeechT5Config) on the WIS side. The default is 600 tokens but mapping characters/words to tokens is tricky. We could investigate increasing that value as well.

nikito, 2023-06-09:
Ok I'll try to bump the value and rebuild/test. The whitespace I managed to get rid of, had some jinja2 templates where I forgot to add the trim characters to strip it between template conditionals, so that's no big deal. Still not sure why that last example I posted failed though, pretty normal string with no extraneous whitespace.

Is the nginx cache in the WIS container? or is it in the nginx one?

kristiankielhofner, 2023-06-09:
The nginx cache is in nginx :). What's really nice about the nginx cache is that WIS doesn't even see the TTS request and we can leverage all of the performance made available by the nginx cache and nginx generally.

In terms of these weird errors, can you provide the output from WIS with `LOG_LEVEL=debug` set in `.env` ?

I'll be doing another deep dive on TTS next week and may be looking for some additional testing...

nikito, 2023-06-09:
So funny thing, I bumped up the hass reply size to 1024, and did my weather question. Worked perfectly with TTS lol. Likewise, I tried the issue above and that also worked fine. But then I tried asking "What lights are on" and got this in my monitor console:
[image]

Looking on WIS, I see this:
[image]

kristiankielhofner, 2023-06-09:
What happens when you `curl`, `wget`, or try to go to that WIS TTS url in a browser?

Also - can you possibly provide this output as text instead of screenshots? It's slow and error-prone for me to read these URLs and output and try to reproduce locally.

nikito, 2023-06-09:
So, I went ahead and deleted the nginx cache, and now it seems fine so far. I tried all the above examples and every one worked perfectly.
So maybe my cache was corrupted somehow?

Also seems the 1024 value for the speech array and HASS reply is good for my examples, though not sure if we should bump this up to a larger value in case larger replies come along (thinking in the future if people start using the chatbot, it may generate long TTS replies)? I can experiment with bumping it up further if you want?

nikito, 2023-06-09:
Entirely possible in my tinkering/building/rebuilding I may have messed something up 😆 I also was running as root originally because I was lazy, and recently went ahead and added my normal user to dialout and docker group so I could do the commands without sudo/root. Maybe that caused these permissions issues I was seeing.

kristiankielhofner, 2023-06-09:
We should probably add a `purge-cache` arg to utils or something that just blows away the cache (for a variety of reasons).

stintel, 2023-06-14:
> In terms of the value for the replies, we should probably attempt to generally align it with the SpeechT5 model configuration max_length. I'd also like to run this by @stintel to make sure there aren't any issues that I may not be considering.

I chose 64 solely on the responses I got testing intents locally. None of them came even close to 63 characters so I figured 64 would be enough. Guess that wasn't a good decision. As for aligning with SpeechT5 ... Maybe we should change it to a pointer and use calloc instead?

stintel, 2023-06-14:
Can you try https://github.com/toverainc/willow/tree/issue/159?

nikito, 2023-06-14:
> Can you try https://github.com/toverainc/willow/tree/issue/159?
Will do, will report back!

nikito, 2023-06-14:
So when I went to build after configure, I get this error:
`FAILED: esp-idf/main/CMakeFiles/spiffs_ui_bin /willow/build/esp-idf/main/CMakeFi                                                                                                             les/spiffs_ui_bin
cd /willow/build/esp-idf/main && /opt/esp/tools/cmake/3.23.1/bin/cmake -E echo "                                                                                                             Failed to create SPIFFS image for partition 'ui'. " "Check project configuration                                                                                                              if using the correct partition table file." && /opt/esp/tools/cmake/3.23.1/bin/                                                                                                             cmake -E remove /willow/build/esp-idf/main/ZcLhu.cmake && /opt/esp/tools/cmake/3                                                                                                             .23.1/bin/cmake -E env "FAIL_MESSAGE=Failing the build (see errors on lines abov                                                                                                             e)" /opt/esp/tools/cmake/3.23.1/bin/cmake -P /opt/esp/idf/tools/cmake/scripts/fa                                                                                                             il.cmake
Failed to create SPIFFS image for partition 'ui'.  Check project configuration i                                                                                                             f using the correct partition table file.
CMake Error at /opt/esp/idf/tools/cmake/scripts/fail.cmake:3 (message):
  Failing the build (see errors on lines above)

[4/1412] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[5/1412] Building C object esp-idf/mbe...y/CMakeFiles/mbedx509.dir/pkcs11.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
root@3ec4087b3cf9:/willow#
`

stintel, 2023-06-14:
I doubt that's related to the issue/159 branch. Try removing the build directory. If that doesn't work, `./utils.sh destroy` and start from scratch.

nikito, 2023-06-14:
Yeah just tried that thinking the same thing and it worked. Will continue and let you know how it goes, thanks!

nikito, 2023-06-14:
Tested with my longest output intent and it worked just fine! 😃

stintel, 2023-06-14:
> Tested with my longest output intent and it worked just fine!

How does the display handle that long text?

nikito, 2023-06-15:
Ran a few tests, it seems fine except that it truncates the last character
of the string. So for instance if the last word is "detection" the display
showed "detection". Likewise if the last part is "degrees." I get "degrees"
(the period is truncated). Otherwise no other anomalies that I noticed. 😊

kristiankielhofner, 2023-06-15:
This is great feedback (it doesn't crash - yeah)!

Generally when it comes to the display it's more-or-less lowest priority (not only in terms of task priority but also functionality). The general use case is assumed to be far-field voice where text on the very small display can't be read from distances of more than a few feet away anyway. Not to mention the relatively slow rate of horizontal scroll, etc. Speaking for myself personally I question the use-case of sitting in front of the device and waiting for the text to scroll from the output of a command - perhaps other than users with hearing issues, which is mostly why it is there in the first place (plus it's kind of cool).

nikito, 2023-06-15:
Agree, I actually had to keep poking the screen to wake it up to see all the text, especially on the paragraph-long weather output 🤣

stintel, 2023-06-15:
> Ran a few tests, it seems fine except that it truncates the last character of the string.

Classic off-by-one bug!

commit 1f0ba485 (2023-07-11): hass: dynamically allocate memory for speech

Closes: #228
Fixes: #159
commit 5ee43c0f (2023-07-11): hass: dynamically allocate memory for speech

Closes: #228
Fixes: #159