HeyWillow/willow #166: Flashing Fails - No Such File

jaswalters, 2023-06-13:
The erase-flash is successful. However, no matter how hard I try the flash always returns
esptool merge_bin: error: argument <address> <filename>: [Errno 2] No such file or directory: 'bootloader/bootloader.bin'

Device: ESP32 Box S3 (non lite version)

hamishcunningham, 2023-06-13:
That suggests that there's a partially successful build that it is trying
to flash but failing. Delete the build completely and try again maybe?

jaswalters, 2023-06-13:
I've tried. I'm also attempting this through a VM which may or may not be
contributing to the issue. I guess I'll wait until we have a Windows
PowerShell script or similar readily available.

jaswalters, 2023-06-14:
I used the dist parameter to create the bin file and transferred to Windows. I was able to flash using the ESP Flash Download Tool.

closed 2023-06-14 by jaswalters