HeyWillow/willow #169: Home Assistant Intents Not Working Well Or At All

vexter0944, 2023-06-14:
I got my ESP BOX (non-lite) in a couple of days ago, got it flashed and up and running pretty quickly.  I tried some intents right after setting it up and none of them would work.  It just returned the I don't understand or similar from HASS when it tried to send the commands.  HASS is Home Assistant 2023.6.1 Frontend 20230608.0 - **latest**

I have a basement area in HASS with a lamp called table_lamp (note the underscore in case it's important) - I sent it a 'turn off basement table lamp' and it didn't understand.  I then set an alias in HASS for the same lamp - 'basement table lamp' - retried and it worked.  

Then I have an entity called shades in my Living Room area - I tried 'open living room shades' - didn't understand.  But when I used the HASS assist and typed it in - it worked perfectly.  

Both times, I made sure that what I said and what it sent were what I expected on the ESP-BOX screen.  It appeared as such.  

Am I missing something here, is this expected currently?  Anything I need to do/check/tweak that I have maybe looked over?

I like the device, it wasn't too bad to flash and setup and looks nice, seems to hear well.  Hoping to understand this issue as I'd love to have a few of these in the house to get back to voice control without Google!

nikito, 2023-06-14:
Are you running with local commands, or WIS? If it is local commands I don't think it supports these types of area commands.

stintel, 2023-06-14:
Try testing your intents in the Home Assistant web interface. Click the text balloon icon in the top right corner, and type your intents there to see if they work.

vexter0944, 2023-06-14:
If you're asking if I have setup the local hosted server - the answer is no.  I plan to, but have not yet.  I'm using the default settings for the ESP BOX (other then my wifi/password and HASS token).  I can work on setting up WIS locally and see how that goes as well.  

If I've missed any info I need to let you all know - let me know.  Thanks for the input and help!

nikito, 2023-06-14:
So there's a tovera hosted instance that is provided by default. But if you are using the literal default config, then it would be set to local by default I think, which is using the ESP local commands, which have a very specific format. You would want to use WIS instead, such as this:
[image]

vexter0944, 2023-06-14:
Here's a screenshot form the config:
[image]

stintel, 2023-06-14:
Rebuild with this change and look at the WebSocket messages for a hint:

```patch
diff --git a/main/endpoint/hass.c b/main/endpoint/hass.c
index 2bc73a2..f71318f 100644
--- a/main/endpoint/hass.c
+++ b/main/endpoint/hass.c
@@ -42,7 +42,7 @@ static void cb_ws_event(const void *arg_evh, const esp_event_base_t *base_ev, co
                 char *json = NULL;
                 char *resp = strndup((char *)data->data_ptr, data->data_len);

-                ESP_LOGD(TAG, "received text data on WebSocket: %s", resp);
+                ESP_LOGI(TAG, "received text data on WebSocket: %s", resp);

                 cJSON *cjson = cJSON_Parse(resp);
                 if (!cJSON_IsObject(cjson)) {
```

Alternatively capture them with Wireshark.

vexter0944, 2023-06-14:
> Hmm ok, so if you say some random string of text I'm guessing it shows on the screen. Not sure why else it wouldn't be recognizing it. Do other commands work such as asking if any lights are on?

I think I tried some other commands, but can test again and report back.  Will be a bit before I can do that.

vexter0944, 2023-06-14:
> Rebuild with this change and look at the WebSocket messages for a hint:
> 
> ```diff
> diff --git a/main/endpoint/hass.c b/main/endpoint/hass.c
> index 2bc73a2..f71318f 100644
> --- a/main/endpoint/hass.c
> +++ b/main/endpoint/hass.c
> @@ -42,7 +42,7 @@ static void cb_ws_event(const void *arg_evh, const esp_event_base_t *base_ev, co
>                  char *json = NULL;
>                  char *resp = strndup((char *)data->data_ptr, data->data_len);
> 
> -                ESP_LOGD(TAG, "received text data on WebSocket: %s", resp);
> +                ESP_LOGI(TAG, "received text data on WebSocket: %s", resp);
> 
>                  cJSON *cjson = cJSON_Parse(resp);
>                  if (!cJSON_IsObject(cjson)) {
> ```
> 
> Alternatively capture them with Wireshark.

Roger - will be a bit before I can try all this, but certainly appreciate the input.  I'll report back in with what I find!

vexter0944, 2023-06-14:
I asked what lights are on and got a list back - so that's good!

stintel, 2023-06-14:
The intent-end event says `no_intent_match`. Are you sure it works if you type exactly `turn on basement table lamp` in Assist in the web interface? I'd find it very strange if that works... Next step then would be to enable debug logging in Home Assistant and check the logs.

vexter0944, 2023-06-14:
> The intent-end event says `no_intent_match`. Are you sure it works if you type exactly `turn on basement table lamp` in Assist in the web interface? I'd find it very strange if that works... Next step then would be to enable debug logging in Home Assistant and check the logs.

No - it does not work on assist when I type  exactly `turn on basement table lamp` - so at least its consistent.  
I'll crank up the logs on HASS and see what I can see there.  

Thanks for the help!

closed 2023-06-16 by kristiankielhofner