HeyWillow/willow #174: Device restarts.

bert269, 2023-06-15:
Hi

I updated to the latest version of the software this morning on ESP_BOX Lite.
The devices keeps rebooting after It issued a command or two (in this case it happened after I turned the light off). 
Here is what I see in TIO:

I (20:50:26.087) WILLOW: app_main() - start_rec() finished
I (20:50:26.090) WILLOW: cmd_multinet[] size: 7486 bytes
I (20:50:26.092) WILLOW: MAC address: 7c:df:a1:ff:38:cc
I (20:50:26.097) WILLOW: Startup complete! Waiting for wake word.
I (20:50:36.104) WILLOW: Wake LCD timeout, turning off LCD
I (20:50:53.044) WILLOW: AUDIO_REC_WAKEUP_START
I (20:50:53.975) WILLOW: AUDIO_REC_VAD_START
I (20:50:57.564) WILLOW: AUDIO_REC_VAD_END
I (20:50:57.565) WILLOW: AUDIO_REC_WAKEUP_END
TOP 1, command_id: 103, phrase_id: 103, prob: 0.524125
I (20:50:58.608) WILLOW: sending command to Home Assistant via WebSocket: {
	"end_stage":	"intent",
	"id":	1686840658,
	"input":	{
		"text":	"Turn on Foyer Light."
	},
	"start_stage":	"intent",
	"type":	"assist_pipeline/run"
}
I (20:50:58.625) WILLOW: Got local command ID: '103'
I (20:50:58.723) WILLOW: home assistant response_type: action_done
I (20:50:58.727) WILLOW: received run-end event on WebSocket: {
	"id":	1686840658,
	"type":	"event",
	"event":	{
		"type":	"run-end",
		"data":	null,
		"timestamp":	"2023-06-15T14:50:58.750327+00:00"
	}
}
I (20:51:09.745) WILLOW: Wake LCD timeout, turning off LCD
I (20:51:20.242) WILLOW: AUDIO_REC_WAKEUP_START
I (20:51:21.811) WILLOW: AUDIO_REC_VAD_START
I (20:51:25.372) WILLOW: AUDIO_REC_VAD_END
I (20:51:25.373) WILLOW: AUDIO_REC_WAKEUP_END
TOP 1, command_id: 104, phrase_id: 104, prob: 0.466570
I (20:51:26.404) WILLOW: sending command to Home Assistant via WebSocket: {
	"end_stage":	"intent",
	"id":	1686840686,
	"input":	{
		"text":	"Turn off Foyer Light."
	},
	"start_stage":	"intent",
	"type":	"assist_pipeline/run"
}
I (20:51:26.419) WILLOW: Got local command ID: '104'
I (20:51:26.506) WILLOW: home assistant response_type: action_done
I (20:51:26.510) WILLOW: received run-end event on WebSocket: {
	"id":	1686840686,
	"type":	"event",
	"event":	{
		"type":	"run-end",
		"data":	null,
		"timestamp":	"2023-06-15T14:51:26.534863+00:00"
	}
}
E (73796) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (73796) task_wdt:  - IDLE (CPU 1)
E (73796) task_wdt: Tasks currently running:
E (73796) task_wdt: CPU 0: IDLE
E (73796) task_wdt: CPU 1: feed_task
E (73796) task_wdt: Aborting.

abort() was called at PC 0x4209c2ec on core 0

Backtrace: 0x4037a18a:0x3fca00d0 0x40383f11:0x3fca00f0 0x4038bfea:0x3fca0110 0x4209c2ec:0x3fca0180 0x4037b1f5:0x3fca01b0 0x42132e63:0x3fcb1eb0 0x42004569:0x3fcb1ed0 0x403852ff:0x3fcb1ef0

ELF file SHA256: 6f07a5fbf5164870

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379e0c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x171c
load:0x403c9700,len:0xc14
load:0x403cc700,len:0x3364
entry 0x403c9968
I (24) boot: ESP-IDF v4.4.4-448-gfd5e03b221-dirty 2nd stage bootloader
I (24) boot: compile time 14:41:50
I (24) boot: chip revision: v0.1
I (28) boot.esp32s3: Boot SPI Speed : 80MHz
I (33) boot.esp32s3: SPI Mode       : DIO
I (37) boot.esp32s3: SPI Flash Size : 16MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00200000
I (81) boot:  3 model            Unknown data     01 82 00210000 00500000
I (88) boot:  4 ui               Unknown data     01 82 00710000 00080000
I (96) boot: End of partition table
I (100) esp_image: segment 0: paddr=00010020 vaddr=3c140020 size=4bf84h (311172) map
I (164) esp_image: segment 1: paddr=0005bfac vaddr=3fc9f5d0 size=0406ch ( 16492) load
I (168) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=1385c4h (1279428) map
I (400) esp_image: segment 3: paddr=001985ec vaddr=3fca363c size=02dc0h ( 11712) load
I (402) esp_image: segment 4: paddr=0019b3b4 vaddr=40378000 size=175d0h ( 95696) load
I (435) boot: Loaded app from partition at offset 0x10000
I (435) boot: Disabling RNG early entropy source...
I (447) opi psram: vendor id : 0x0d (AP)
I (447) opi psram: dev id    : 0x02 (generation 3)
I (447) opi psram: density   : 0x03 (64 Mbit)
I (450) opi psram: good-die  : 0x01 (Pass)
I (455) opi psram: Latency   : 0x01 (Fixed)
I (460) opi psram: VCC       : 0x01 (3V)
I (465) opi psram: SRF       : 0x01 (Fast Refresh)
I (470) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (476) opi psram: BurstLen  : 0x01 (32 Byte)
I (481) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (487) opi psram: DriveStrength: 0x00 (1/1)
I (492) spiram: Found 64MBit SPI RAM device
I (496) spiram: SPI RAM mode: sram 80m
I (501) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (508) cpu_start: Pro cpu up.
I (512) cpu_start: Starting app cpu, entry point is 0x403796c4
I (500) cpu_start: App cpu up.
I (803) spiram: SPI SRAM memory test OK
I (812) cpu_start: Pro cpu start user code
I (812) cpu_start: cpu freq: 240000000
I (812) cpu_start: Application information:
I (815) cpu_start: Project name:     willow
I (820) cpu_start: App version:      1
I (824) cpu_start: Compile time:     Jun 15 2023 14:41:41
I (830) cpu_start: ELF file SHA256:  6f07a5fbf5164870...
I (836) cpu_start: ESP-IDF:          v4.4.4-448-gfd5e03b221-dirty
I (843) cpu_start: Min chip rev:     v0.0
I (848) cpu_start: Max chip rev:     v0.99 
I (853) cpu_start: Chip rev:         v0.1
I (857) heap_init: Initializing. RAM available for dynamic allocation:
I (865) heap_init: At 3FCAB250 len 0003E4C0 (249 KiB): D/IRAM
I (871) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (878) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (884) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (892) spi_flash: detected chip: gd
I (896) spi_flash: flash io: dio
I (901) sleep: Configure to isolate all GPIO pins in sleep state
I (907) sleep: Enable automatic switching of GPIO sleep configuration
I (914) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (929) spiram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (14:51:31.599) WILLOW: Starting up! Please wait...
I (14:51:31.729) WILLOW: Waiting on SPIFFS mount...
I (14:51:32.230) WILLOW: SPIFFS mounted
W (14:51:36.626) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (14:51:37.626) WILLOW: initializing SNTP client
I (20:51:37.627) WILLOW: Using configured SNTP server 'pool.ntp.org'
I (20:51:37.653) WILLOW: audio_hal_ctrl_codec: ESP_OK
I (20:51:37.654) WILLOW: HASS URL: http://192.168.1.233:8123/api/components
I (20:51:37.683) WILLOW: HTTP status='200' content_length='4708'
I (20:51:37.688) WILLOW: Home Assistant has Assist Pipeline support
I (20:51:37.690) WILLOW: HASS URL: ws://192.168.1.233:8123/api/websocket
I (20:51:37.706) WILLOW: WS event ID: 1
E (20:51:37.710) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(128): i2c transaction failed
E (20:51:37.711) TT21100: esp_lcd_touch_tt21100_read_data(173): I2C read error!
E (20:51:37.716) TT21100: esp_lcd_touch_new_i2c_tt21100(103): TT21100 init failed
E (20:51:37.725) TT21100: Error (0xffffffff)! Touch controller TT21100 initialization failed!
E (20:51:37.734) WILLOW: failed to initialize touch screen: ESP_FAIL

----------------------------- ESP Audio Platform -----------------------------
|                                                                            |
|                       ESP_AUDIO-v1.7.2-20e6bd0-b92a149                     |
|                     Compile date: Nov 30 2022-07:50:12                     |
------------------------------------------------------------------------------
I (20:51:37.779) WILLOW: audio player initialized
E (20:51:37.781) I2S: register I2S object to platform failed
I (20:51:37.789) WILLOW: Using local multinet
MC Quantized wakenet9: wakenet9l_v3h24_alexa_3_0.625_0.645, tigger:v3, mode:5, p:0, (May  5 2023 20:32:52)
Build fst from commands.
Quantized MultiNet6:rnnt_ctc_1.0, name:mn6_en, (May  5 2023 20:32:52)
Load 292 pre-defined speech commands.
Quantized MultiNet6 search method: 2, time out:2.0 s
I (20:51:39.710) WILLOW: app_main() - start_rec() finished
I (20:51:39.712) WILLOW: cmd_multinet[] size: 7486 bytes
I (20:51:39.714) WILLOW: MAC address: 7c:df:a1:ff:38:cc
I (20:51:39.720) WILLOW: Startup complete! Waiting for wake word.
I (20:51:40.889) WILLOW: SNTP client synchronized time to 1686840700
I (20:51:49.925) WILLOW: Wake LCD timeout, turning off LCD

I will try the ESP-BOXs in a little an report back if they behave the same as the LITE.
Thanks

kristiankielhofner, 2023-06-15:
We have a branch with improved multinet. Can you try it?

```
git checkout feature/multinet-text-results
git pull
./utils.sh reinstall
./utils.sh build
```

Then flash and let us know!

kristiankielhofner, 2023-06-15:
Great, it's in the branch for more testing but this is a very good sign that we will merge it into main soon (where it will live forever)!

Thanks again for testing this, I'll leave this issue open in case you or others have additional issues.

bert269, 2023-06-15:
This works fine also on ESP-BOXs.
Thanks.

kristiankielhofner, 2023-06-16:
This has been merged into main.

closed 2023-06-16 by kristiankielhofner

bert269, 2023-06-17:
I see you clsoed this issue.
I received another ESPBOX yesterday afternnon and flashed it yesterday evening, but it keeps on rebooting also.- same messages as I received originally. I will wipe the docker environment, redownload, rebuild and reflash to ensure I pick this "merge" up - will report back in a few moments.

kristiankielhofner, 2023-06-17:
Very strange. Did you just re-run the ```./utils.sh flash``` command you used to flash the working device, or did you take any other config/build steps in between the unit that works and the one that doesn't?

We routinely flash multiple devices with successive flash commands and the results across devices are as you would expect - identical.