HeyWillow/willow #176: Having an issue with an ESP32-S3-BOX restarting continuously after mounting SPIFFS

Zectivi, 2023-06-15:
Hello!

Two days ago, I was able to get Willow installed on this device. Since then, it worked great! Last night, I issued a command to shut off the lights, and while the lights turned off, the screen remained with "Say Command...".

This morning, I noticed the screen was flashing. It briefly shows "Welcome to Willow! Starting up (server) Text" and it goes black, then white again with the same text.

I've attempted a `flash` or an `erase-flash` -> `flash`  a few times, including doing a `destroy` and setup from the start. It still does it.

I also have an ESP32-S3-BOX-LITE available that was brand new out of box, so I configured and flashed that, and it's working great at the moment.

After the most recent flash, it gets as far (via Tio) as saying "WILLOW: SPIFFS mounted" then it Disconnects.

Here are the flash logs:

```
$ sudo PORT=/dev/ttyACM0 ./utils.sh flash
Using venv for esptool
esptool esp32s3 -p /dev/ttyACM0 -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 ui.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:fd:17:b8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x001b1fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00210000 to 0x00681fff...
Flash will be erased from 0x00710000 to 0x0078ffff...
Compressed 22256 bytes to 13957...
Wrote 22256 bytes (13957 compressed) at 0x00000000 in 0.2 seconds (effective 812.3 kbit/s)...
Hash of data verified.
Compressed 1711488 bytes to 1103965...
Wrote 1711488 bytes (1103965 compressed) at 0x00010000 in 10.4 seconds (effective 1310.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 131...
Wrote 3072 bytes (131 compressed) at 0x00008000 in 0.0 seconds (effective 688.8 kbit/s)...
Hash of data verified.
Compressed 4660656 bytes to 3597428...
Wrote 4660656 bytes (3597428 compressed) at 0x00210000 in 30.7 seconds (effective 1213.7 kbit/s)...
Hash of data verified.
Compressed 524288 bytes to 68519...
Wrote 524288 bytes (68519 compressed) at 0x00710000 in 1.8 seconds (effective 2300.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[12:49:57.109] tio v2.6
[12:49:57.109] Press ctrl-t q to quit
[12:49:57.110] Connected
I (100) esp_image: segment 0: paddr=00010020 vaddr=3c1400I (166) esp_image: segment 1: paddr=0005abd4 vaddr=3fc9f690 size=05444h ( 21572) load
I (171) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=139110h (1282320) map
I (402) esp_image: segment 3: paddr=00199138 vaddr=3fca4ad4 size=01580h (  5504) load
I (403) esp_image: segment 4: paddr=0019a6c0 vaddr=40378000 size=17690h ( 95888) load
I (437) boot: Loaded app from partition at offset 0x10000
I (437) boot: Disabling RNG early entropy source...
I (449) opi psram: vendor id : 0x0d (AP)
I (449) opi psram: dev id    : 0x02 (generation 3)
I (449) opi psram: density   : 0x03 (64 Mbit)
I (452) opi psram: good-die  : 0x01 (Pass)
I (457) opi psram: Latency   : 0x01 (Fixed)
I (462) opi psram: VCC       : 0x01 (3V)
I (467) opi psram: SRF       : 0x01 (Fast Refresh)
I (472) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (478) opi psram: BurstLen  : 0x01 (32 Byte)
I (483) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (489) opi psram: DriveStrength: 0x00 (1/1)
I (494) MSPI Timing: PSRAM timing tuning index: 4
I (499) spiram: Found 64MBit SPI RAM device
I (504) spiram: SPI RAM mode: sram 80m
I (508) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (515) cpu_start: Pro cpu up.
I (519) cpu_start: Starting app cpu, entry point is 0x403796d0
I (0) cpu_start: App cpu up.
I (808) spiram: SPI SRAM memory test OK
I (816) cpu_start: Pro cpu start user code
I (816) cpu_start: cpu freq: 240000000
I (816) cpu_start: Application information:
I (819) cpu_start: Project name:     willow
I (824) cpu_start: App version:      1
I (828) cpu_start: Compile time:     Jun 15 2023 17:45:38
I (834) cpu_start: ELF file SHA256:  6354e7f9b45bc3dc...
I (840) cpu_start: ESP-IDF:          v4.4.5-98-ge33a1808d6-dirty
I (847) cpu_start: Min chip rev:     v0.0
I (852) cpu_start: Max chip rev:     v0.99 
I (857) cpu_start: Chip rev:         v0.1
I (861) heap_init: Initializing. RAM available for dynamic allocation:
I (869) heap_init: At 3FCAAF30 len 0003E7E0 (249 KiB): D/IRAM
I (875) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DIRAM
I (882) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (888) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (896) spi_flash: detected chip: gd
I (900) spi_flash: flash io: dio
I (905) sleep: Configure to isolate all GPIO pins in sleep state
I (911) sleep: Enable automatic switching of GPIO sleep configuration
I (918) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (933) spiram: Reserving pool of 16K of internal memory for DMA/internal allocations
I (00:00:00.120) WILLOW: Starting up! Please wait...
I (00:00:00.251) WILLOW: Waiting on SPIFFS mount...
I (00:00:00.751) WILLOW: SPIFFS mounted

[12:50:00.943] Disconnected
[12:50:01.945] Connected
I (400) esp_image: segment 3: paddr=00199138 vaddr=3fca4aI (00:00:00.251) WILLOW: Waiting on SPIFFS mount...
I (00:00:00.751) WILLOW: SPIFFS mounted

[12:50:04.889] Disconnected
[12:50:05.891] Connected
I (400) esp_image: segment 3: paddr=00199138 vaddr=3fca4aI (00:00:00.251) WILLOW: Waiting on SPIFFS mount...
I (00:00:00.751) WILLOW: SPIFFS mounted

[12:50:08.835] Disconnected
```
If there is anything else I could try, or other logs, please let me know what and where I need to get! Thank you so much for this project and any potential help provided!

stintel, 2023-06-22:
Can you do a build with bogus Wi-Fi and endpoint credentials, verify that you still experience the same problem, and then do a full flash dump with esptool read_flash, upload the image somewhere and pass us a link so we can try to reproduce this locally?

Zectivi, 2023-06-22:
Yes. I'll work on this later today after work.

I have since the initial report tried a different cable on a different USB port, and I flashed the device back to the expressif image and cleanly pulled/built/flashed Willow and have the same issue. 

Also happy to report that the BOX-Lite I also flashed is going strong, so this starts to maybe feel like a bad unit, but I'll get that image up here soon.

stintel, 2023-06-22:
My gut feeling says you have a bad flash chip.

Zectivi, 2023-06-22:
Changed config for wifi and HA url/token. Built, flashed.
Command executed: `esptool.py --port /dev/ttyACM0 read_flash 0 ALL zectivi-read_flash.bin`

File via pCloud share: https://u.pcloud.link/publink/show?code=XZ7mbUVZX2owfVuEDS4dqWB4pdWTjz3S2IvX

stintel, 2023-06-22:
I flashed your image on an ESP32-S3-Box and set up an SSID with the Wi-Fi credentials you configured in your build. It boots fine. You mention you tried a different USB port, I assume on your computer. Have you tried powering the ESP32-S3-Box with a decent USB power supply rather than off your computer's USB port? You won't be able to see the serial log, but you should be able to detect a boot loop from the display resetting...

Zectivi, 2023-06-23:
Good call on trying another power source.

I tried various USB power adapters. The only one that seemed to have any effect was the USB-C adapter for my laptop. It was able to keep the screen on for about 1 second longer before restarting, so I gave it a little time. Eventually and on a couple occasions of allowing it to boot loop, it stayed up and attempted to connect to wifi. That doesn't happen when connected to other power sources. I tried both the dock and the direct power port and both did the same thing

Then I had the idea that if I can boot it on the laptop power, what would it do if I connected a second source directly to the box and unplugged the laptop. Well, it stayed up for about 30 seconds, and then began boot looping again.

At this point, it seems very much like a hardware issue and will look at doing an RMA or return for a new unit. 

Thank you so much @stintel for the assist and helping to figure it out. I will go ahead and close since it is not a Willow issue.

closed 2023-06-23 by Zectivi

kristiankielhofner, 2023-06-23:
It certainly does seem like a hardware issue and to my knowledge this is the first hardware issue we're hearing of with the ESP BOX. I've always been impressed with the reliability and robustness of Espressif devices but even they ship a lemon every once in a while!