HeyWillow/willow #182: Device identification in Home Assistant

A6blpka, 2023-06-17:
First of all, thank you for a great project! I've been playing with esp-box for the third day now, and it's great!

What the Home Assistant is missing:

1. Identifying the device from which the command came. What is this for? With my current voice assistant, I understand which room I am giving a command to, and all I have to say is "Turn on the lights" or "Lights" to control the lights in the room where the device is.
Maybe this can be solved by implementing a custom component in HA? I'll try to look into the HA documentation, a quick look at the HA source code shows that "ConversationInput" contains the "device_id" property.

2. It's pretty important to understand Willow's current state in the HA. When the current voice assistant goes to the "Listening" or "Speaking" state, I mute the TV, for example, so that it does not interfere with the assistant's communication, and then I put sound  back. I understand that this refers to the integration of Willow itself with Home Assistant, but since Willow works not only with HA, but also with openHAB, it is probably worth looking at mqtt, as a generic interface.

kristiankielhofner, 2023-06-17:
Thanks!

1. See #74. With this we will only wake on one device (ideally the "closest" one by audio level). Between that and dynamic configuration in Willow 1.0 (early July) we will support the ability to not only activate Willow closest to the speaker but also the ability to configure location (likely by pulling zones from HA when used with HA). Note that location will be post 1.0 functionality.

2. Tighter integration with HA is something many users would appreciate and we could provide this functionality with the appropriate signaling support (as you note) - it would also likely help with 1 noted above. However, as you point out our goal is to be command platform agnostic.

To my knowledge HA has no ability to provide this additional data via API (or anything other than output text). It may be possible down the road with a Willow Home Assistant component. This is actually where we are very interested in working with the community. Our internal development resources (@stintel and I) have our hands pretty full just between Willow, Willow Inference Server, and Willow Application Server (in development). We would have to look hard at how we would implement this from a signaling standpoint in a way that has the potential to work with multiple command platforms. This is non-trivial and a non-starter prior to an HA component anyway...

lordratner, 2023-06-19:
From what I've seen over the years, most HA integrations end up being made by savvy users who really want some service to work with HA, so they make it happen. 

With what I've seen so far with WIS and Willow, once you have to 1.0 out and an easy setup, this should spread like wildfire over there, and pull in some people to help. Self-hosting voice assistants is one of the holy grails of home automation, and with HA limited to the Raspberry Pi being their primary hardware, I don't suspect they will ever be able to match WIS. That should lead users looking for more (especially the possibility of LLM support!) to pitch in.

kristiankielhofner, 2023-06-19:
That aligns with what I've seen from the Home Assistant community too and you make a good point - the cumbersome install instructions make Willow and non-starter for many people and we should get more attention from the HA community once install/flash/config is a few clicks.

A6blpka, 2023-09-14:
Relates to https://github.com/home-assistant/core/pull/99363.