HeyWillow/willow #188: Timezone setting working in reverse

blakadder, 2023-06-17:
When setting UTC+2 as timezone in config my time becomes -2 from UTC and when setting timezone as UTC-2 the time is 2 hours ahead of UTC which is my actual local time (CEST or UTC+2)

kristiankielhofner, 2023-06-18:
This is strange but noticeable even from the default "US Central time-ish".

We need to look at this because even with dynamic configuration we should be able to pass offset as we do today and have it apply correctly.

stintel, 2023-06-18:
See https://github.com/toverainc/willow/pull/85#issuecomment-1556359299 and https://github.com/toverainc/willow/pull/85#issuecomment-1557333771. Most likely you are just setting an incorrect timezone.  You should use `CET-1CEST,M3.5.0,M10.5.0/3` for Croatia.

We will expose timezone in the form of Continent/City in WAS, and convert those to the correct POSIX timezone string in the config, similar to #85.

closed 2023-06-18 by stintel