HeyWillow/willow #193: Parameters for dynamic configuration

kristiankielhofner, 2023-06-19:
The dynamic configuration manifest should also include:

- Hardware device type (ESP BOX, ESP BOX Lite)
- A user label and/or description

lordratner, 2023-06-20:
Static IP/Gateway/DNS options would be handy. I didn't see that in the current config.

lordratner, 2023-06-20:
> Static IP/Gateway/DNS options would be handy. I didn't see that in the current config.

Nevermind, it looks like you already have it from the screenshot of WAS in another post. Cheers!

kristiankielhofner, 2023-06-20:
Yes, but it doesn't support static IP. Binding a MAC address to a specific IP address will have to be done in your router/DHCP server for now.

stintel, 2023-06-26:
Device type was added in c02b457d915d25da88773c7d2d6b2422d4c59d0d.

stintel, 2023-08-17:
Once we merge https://github.com/toverainc/willow-application-server/pull/15 this can be closed.

closed 2023-09-05 by stintel