HeyWillow/willow #201: I need to be fairly close to the device for it to "hear" me

mhilbush, 2023-06-23:
I have two devices running Willow built from a repo I cloned on June 11. Each device is in a completely separate part of the house (1st floor kitchen and lower level family/rec room).

I only ever use the wake word "Alexa".

I find that I need to speak pretty loudly even when I’m relatively close to the device (5-6 feet). There's very little to no background noise in the areas where the devices are located.

Please see issue #199 for a description and photos of the environment where my devices are located.

kristiankielhofner, 2023-06-24:
In the configuration step of the build, go to the Advanced settings menu -> Wake Word Recognition Operating Mode and select 'DET_MODE_2CH_95'. If you're feeling very adventurous you can also try 3 channel 95.

That increases the "aggressiveness" of wake word detection.

By default we have extremely conservative settings and it's feedback and testing like this that helps us update our defaults to serve more users better!

mhilbush, 2023-06-24:
Running with `DET_MODE_2CH_95`, this morning I had to say "Alexa" 3 times before it woke up.  I was standing at the kitchen counter directly in front of the device. I'll try 3CH_95 to see if that makes a difference.

I noticed in the config that microphone gain is set to the max. I don't remember exactly what it was; something like 35 dB. I assume that's a +dB value? Is that what it should be?

mhilbush, 2023-06-24:
I should add that after the first "Alexa", I tried to not alter the tone, volume, or pace of subsequent "Alexa"s. Of course, I'm not sure how well I actually did that since I was conscious that the first "Alexa" didn't wake it.

mhilbush, 2023-06-26:
Thanks. I'll do a build with save_audio=true on the URL. I'll flash both devices (I had only done 1 so far) and try to get a little more time with them. My daughter was visiting from out of town this w/e so I didn't have much time to spend testing.

kristiankielhofner, 2023-06-26:
No worries! Much to our (very pleasant) surprise we haven't had nearly as many fundamental audio issues as we were expecting. We take them very seriously (of course) so anything you can do to help track this down would be very appreciated!

stintel, 2023-08-17:
Please try https://github.com/toverainc/willow/releases/tag/0.1.0-rc.1.