HeyWillow/willow #202: Crash while making request to WIS

robbles, 2023-06-24:
I got this error and the device rebooted (successfully) afterwards. It appeared to happen while it was processing the input audio.

```
I (12:02:06.269) WILLOW: AUDIO_REC_WAKEUP_START
I (12:02:07.792) WILLOW: AUDIO_REC_VAD_START
E (12:02:07.793) AUDIO_ELEMENT: [http_stream_writer] Element already stopped
E (12:02:07.795) AUDIO_ELEMENT: [raw_stream_writer_to_api] Element already stopped
I (12:02:07.804) WILLOW: Using WIS URL 'https://infer.tovera.io/api/willow'
I (12:02:07.813) WILLOW: WIS HTTP client starting stream, waiting for end of speech
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40056f17  PS      : 0x00060930  A0      : 0x820db5ce  A1      : 0x3d9a1360
A2      : 0xfa25752d  A3      : 0xf917eee4  A4      : 0x0130cf14  A5      : 0xfa25752d
A6      : 0x3d9a13b0  A7      : 0x00000000  A8      : 0x820dd1df  A9      : 0x3d9a1330
A10     : 0x00000020  A11     : 0x3fca313c  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x3d9a13b0  A15     : 0x00000004  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0xf917eee4  LBEG    : 0x42135c5d  LEND    : 0x42135c68  LCOUNT  : 0x00000000

Backtrace: 0x40056f14:0x3d9a1360 |<-CORRUPTED

ELF file SHA256: 41b2bfb64cd960a0

Rebooting...
ESP-ROM:esp32s3-20210327
```

kristiankielhofner, 2023-06-24:
We had some downtime for an upgrade tonight on the best effort server. Have you repeated this?

robbles, 2023-06-24:
No problems since then, it was just a one-off.

I haven't looked at the HTTP client code - is it fairly dependent on the WIS server returning a reasonable response at this point?

kristiankielhofner, 2023-06-24:
We have an open issue that we are in the process of addressing before our first stable release:

#25

closed 2023-06-24 by robbles