HeyWillow/willow #203: Getting error about nvs when running utils dist on WAS branch

nikito, 2023-06-26:
Getting the following when running dist for willow WAS:

nbento@docker-server-1:/willow$ ./utils.sh dist
Willow build version: 3396466-dirty
Using venv for esptool
esptool esp32s3 merge_bin -o /willow/build/willow-dist.bin --flash_m
ode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader
.bin 0x30000 willow.bin 0x8000 partition_table/partition-table.bin 0
x2d000 ota_data_initial.bin 0x630000 srmodels/srmodels.bin 0xc30000
user.bin
esptool.py v4.5.1
Wrote 0x1000000 bytes to file /willow/build/willow-dist.bin, ready t
o flash to offset 0x0
Combined firmware image for flashing written
-rw-rw-r-- 1 nbento nbento 16M Jun 26 00:49 /willow/build/willow-dis
t.bin
./utils.sh: line 174: /opt/esp/idf/components/nvs_flash/nvs_partitio
n_generator/nvs_partition_gen.py: No such file or directory

nikito, 2023-06-26:
Just an update, I think the above is an issue with venv. If I run the dist command inside the docker container it works fine, but if I run it outside it fails. Likewise if I try to use the flash utils after doing dist in the container, it fails outside unless I delete the venv directory and try the erase-flash or flash-dist command again, which then regenerates venv and works, but then breaks commands inside the container 😆

kristiankielhofner, 2023-06-26:
To generate the partition you need the full ESP-IDF framework installed (as it is with the container) or possibly just the nvs_partition_gen.py script but we haven't explored that path.

With WAS and dynamic config approaching anything container/build/config/dev related will be used near exclusively for active development on Willow and we will be doing a massive documentation update with dedicated sections for development in an attempt to prevent or at least explain these kinds of issues.

@stintel knows what works for us specifically - we (of course) have development down to mechanisms that works for us (we're actually quite different in this regard) but we'd certainly be interested to hear from Willow devs such as yourself what your preferred workflow looks like so we can support and document it.

In general, both flash and flash-app don't overwrite NVS so if you have a valid NVS partition already flashed on the device you can build, update, etc without worrying about the NVS generation/dist steps.

closed 2023-09-01 by kristiankielhofner