HeyWillow/willow #208: Devices stop responding to wake word after being idle for a prolonged period of time

nikito, 2023-06-28:
I noticed this morning, after my ESPs were idle overnight, that none of them would respond to the wake phrase (Hi ESP) and would not react to touchscreen presses. I had to power cycle the devices for them to become responsive again. Not sure what logs I could try to gather if it happens again?

Also I am running the WAS branch including the WAS server, and noticed the devices still seemed to be connected to WAS, so at least some piece of it was still alive.

lgcnbento, 2023-06-28:
Related to #199

nikito, 2023-06-28:
Related to #199

stintel, 2023-08-17:
Please try https://github.com/toverainc/willow/releases/tag/0.1.0-rc.1.

nikito, 2023-08-17:
Closing as this has not occured since using latest build. 🙂

closed 2023-08-17 by nikito