HeyWillow/willow #216: AUDIO_THREAD: Not found right ESP hot waking

ardenking, 2023-06-30:
HI there 

Got my SP32 S3 today tested it with shipped code and it woke fine 

Loaded the willow code but now fails to wake and can see this in console 

guess I'm missing something in config but not sure what 

E (12:11:02.293) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12:11:02.303) AUDIO_THREAD: Error creating RestrictedPinnedToCore media_task
E (8560) ESP_AUDIO_CTRL: Error create media_task
E (12:11:02.316) WILLOW: failed to add HTTP input stream to ESP Audio
E (12:11:02.323) WILLOW: failed to add input stream to ESP Audio
E (12:11:02.330) WILLOW: failed to add FLAC decoder to ESP Audio
E (12:11:02.339) WILLOW: failed to add output stream to ESP Audio
E (8592) ESP_AUDIO_CTRL: Error input parameter. line:1163
I (12:11:02.350) WILLOW: audio player initialized
E (12:11:02.355) I2S: register I2S object to platform failed
I (12:11:02.363) WILLOW: Using local multinet
MC Quantized wakenet9: wakeNet9_v1h24_hiesp_3_0.63_0.635, tigger:v3, mode:2, p:0, (Jun 14 2023 11:15:21)
Build fst from commands.
I (12:11:07.209) WILLOW: SNTP client synchronized time to 1688145067
Quantized MultiNet6:rnnt_ctc_1.0, name:mn6_en, (Jun 14 2023 11:15:20)
Load 374 pre-defined speech commands.
Quantized MultiNet6 search method: 2, time out:2.0 s
E (12:11:09.616) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12:11:09.629) AUDIO_THREAD: Error creating RestrictedPinnedToCore feed_task
E (12:11:09.637) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12:11:09.658) AUDIO_THREAD: Error creating RestrictedPinnedToCore fetch_task
E (12:11:09.666) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12:11:09.686) AUDIO_THREAD: Error creating RestrictedPinnedToCore recorder_task
E (12:11:09.694) AUDIO_RECORDER: Fail to create recorder_task
I (12:11:09.703) WILLOW: app_main() - start_rec() finished
E (12:11:09.707) AUDIO_THREAD: Not found right xTaskCreateRestrictedPinnedToCore.
Please enter IDF-PATH with "cd $IDF_PATH" and apply the IDF patch with "git apply $ADF_PATH/idf_patches/idf_v4.4_freertos.patch" first

E (12:11:09.727) AUDIO_THREAD: Error creating RestrictedPinnedToCore at_read
I (12:11:09.736) WILLOW: cmd_multinet[] size: 12027 bytes
I (12:11:09.741) WILLOW: MAC address: f4:12:fa:e2:30:98
I (12:11:09.747) WILLOW: Startup complete! Waiting for wake word.
I (12:11:19.796) WILLOW: Wake LCD timeout, turning off LCD
I (12:13:27.181) WILLOW: Wake LCD timeout, turning off LCD
W (12:17:48.951) PERIPH_WIFI: Wi-Fi disconnected from SSID TALKTALK189FBD-2G, auto-reconnect enabled, reconnect after 1000 ms
E (12:17:48.953) TRANSPORT_BASE: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (12:17:48.964) WEBSOCKET_CLIENT: Network error: esp_transport_poll_read() returned -1, errno=119
I (12:17:48.974) WILLOW: WS event ID: 2
W (12:17:50.009) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (12:17:58.992) WILLOW: WS event ID: 1
I (12:18:09.014) WILLOW: WS event ID: 4
I (12:21:06.000) WILLOW: Wake LCD timeout, turning off LCD

nikito, 2023-06-30:
What wake word did you choose? Do you have a screenshot of settings (with sensitive stuff censored of course :) ) I've got a couple devices flashed several times each and haven't run into this 😮

hamishcunningham, 2023-06-30:
How did you build? The standard method applies an adf patch I think, which
the error suggests you may be missing?

kristiankielhofner, 2023-06-30:
The Dockerfile for our container applies this necessary patch.

You could review it and try to apply the patch manually but the real issue here is how you got this far with that step missing. I'm not sure how you installed, configured, built, and flashed but if the patch didn't apply you likely have other issues.

You should probably start over and make sure you follow the instructions in the README exactly.

ardenking, 2023-07-06:
OK Ive fixed this issue I was using a raspberry pi for the docker as that was what was in my bench at the time 
spun a debian on an old laptop and it worked first time

closed 2023-07-06 by ardenking