HeyWillow/willow #218: Install : Remote branch not found in upstream origin error

XRlabTech, 2023-07-01:
Hi There,

So while attempting to install and run sudo ./utils.sh install 
I received the error below.

Cloning into 'esp-adf'...
fatal: Remote branch not found in upstream origin

I tried to open the ./utils .sh and edit the remote branch but it seemed to cause more issues.

nikito, 2023-07-01:
The install command should be run in the container, and at that point you are root on the container so you shouldn't need to add sudo. Just making sure you are running this from within the container after doing ./utils.sh build-docker and then ./utils.sh docker? Also to avoid the need to use sudo you can add your user to both the dialout group and the docker group. 🙂

stintel, 2023-07-01:
Sounds like you need to rebuild the container. We pointed to the willow branch of our esp-adf fork. If you did git pull without rebuilding the container you will try to get the 2.5 tag of esp-adf, which does not exist in our fork.

stintel, 2023-08-17:
Pretty sure container rebuild should fix this as explained in previous comment. Also with 0.1.0-rc.1 out there, most people will no longer need to use the build container.

closed 2023-08-17 by stintel