HeyWillow/willow #219: willow interface with WIS

bert269, 2023-07-01:
I have setup a WIS on an older AMD motherboard with a GTX 1080ti. The response from willow to the WIS is super fast - but....not accurate.
When I use the exact same configuration on willow for the ESPBOX and for instance want to control the "Foyer Light" or "Foyer Lamp" it respond back with "Fire light or "Fire lamp". Half the time it is an 'unrecognized command".
Also, I have lamps on both sides of my bed, defined in HA as a group "bedsides" - when connected to WIS it either sees it a "bed side" or "beside" - but I cannot get it to recognize "bedsides". When I revert back to Multinet, all the above problems goes away.
I have even tried different settings for VAD timeout (reduced it when I went to WIS) and DET_MODE_3CH-95 changed to 3CH_90 - but it still does not make any change.

Any ideas what settings I can try to get a better accuracy with WIS?
Or is is just my luck again?

Thanks

nikito, 2023-07-01:
You can add ?model=large&beam_size=5 to your wis url in your config to make the model more precise. Given you are using a 1080 ti you should still easily get sub 1 second inference but have higher accuracy. 🙂

bert269, 2023-07-01:
Just to be clear: the wis url should be:  https://192.168.1.239:19000/api/willow?model=large&beam_size=5
The TTS URL I leave the same (https://192.168.1.239:19000/api/tts) ?

Will try and report back.

Nope - does not make any difference at all. Still get the wrong responses: "Fire Light" and now...wait for it...... "both sides"

bert269, 2023-07-01:
It almost seems to me that it has something to do with the volume.
When I raise my voice about 40% louder - it picks up the right command. I realized that "bedsides' are more prune to errors that "bedside". So I renamed the group in HA. 
But now I got it to work now with WIS. Although, the neighbors probably think I hate ALEXA as I am almost shouting at her.

No other changes made to the audio/mic willow settings:
PCM, DET_MODE_3CH_95, VAD_MODE_3, Mic db: 37.5dB

Any other ideas?

bert269, 2023-07-01:
neah - I think it is volume related.
I got it to work now with bedsides, but saying it louder.

bert269, 2023-07-01:
I give up - there is no consistency in the accuracy of these commands. It does not make sense.
My only solution is to define device-groups in HA. For "Foyer Light" and "Foyer Lamp" I created "Fire Light" and "Fire Lamp" groups, which point to the respective device. I did the same for "bedsides" - I greated groups called "Both Sides", "Bedside" and "Bed side".
I do not understand why it works perfectly with Multinet, but not with WIS.
One should expect better accuracy, right, as it does not go out on the WWW?

nikito, 2023-07-01:
Multinet is a restricted model that contains only your devices turned into on/off commands, whereas WIS isn't looking at your devices and is instead simply translating your speech into text with zero context. I've noticed sometimes speaking a bit slower with some of these less common words can help it.

adamast0r, 2023-07-02:
I have noticed this problem before, there is definitely an accuracy problem....I had HA entities that were called "smart plug" and I was having issues, I end up renaming my entities for just one word like "plug" and apparently that works a bit better.

Assuming that we cannot improve the accuracy, maybe we could have an hybrid approach in WIS when we want to turn ON/OFF devices. We could send some context (names of the devices in HA) and similar names would still trigger them....

kristiankielhofner, 2023-07-03:
I'm not sure exactly what's going on with your really poor (nearly random results).

A lot of this comes down to an NLP/NLU approach. The problem (as you note) with general STT is that the responses more-or-less have to be 100% accurate to match with Home Assistant. HA does a little bit of "fuzzy" matching but not nearly enough.

As mentioned, Multinet works because it has predefined commands and just returns a probability that your speech matched one of them (and it's often pretty low if you look at the console). It then sends the exact pre-defined text to Home Assistant.

With the Willow Application Server we will be adding a much more capable NLU/NLP engine that will bring significantly better "fuzzy" matching for grammar and user intent to HA command - the best blend between free-form speech input with STT and probability matching to a command. Of course this doesn't address one of the fundamental issues you seem to be having (straight up garbage STT) but it will greatly help for everything else once the issue you're having gets sorted. One of the fundamental issues here is that Whisper kind of isn't intended for very short commands - it gets the WER scores it does with longer speech segments because it has context. When you feed it a few words (single seconds of audio) it has significantly less to go on to generate a sensical response. We've been looking at other approaches for this as well.

On that note, what versions of Willow and WIS are you running? Have you tried multiple ESP BOX devices? You're experience is pretty far from typical and I'd really like to investigate this further as we prepare for our first stable release.

bert269, 2023-07-05:
Kristian

Thanks for the response. YEs, I have tried this with three ESP-Boxes and
one ESP-Box=lite.
I have tried it with the latest version of the software that was
available as of last Saturday.

Are the commands I can issue to show the versions of willow and WIS?

Thanks

kristiankielhofner, 2023-07-06:
The best way (currently) to get "version" for both Willow and WIS is to execute `git describe --always --dirty --tags` in root of the Willow and WIS directories.

bert269, 2023-07-19:
I had to rebuild the "willows" ubuntu machine I used before to flash ESPBOX (and -lite). Everything is rebuilded, configure willow and flash the ESP-BOX-lite. When I run it, I get these errors, constanly:
[image]

I redownloaded willow and went through all the steps and then flashed again - same problem.

This is my config as it was before:
[image]

Do I miss something on the new "willow/ubuntu" machine side?

Or did something change on the source (Github) side?

bert269, 2023-07-19:
I missed saying that it's the same Ubuntu distro I used before.

nikito, 2023-07-19:
Is WIS up and running? Are you able to reach /api/docs on the WIS instance from a browser?

bert269, 2023-07-19:
That was it. When I did a docker ps - the container was up and running, but had to stop and rm it. Restarted it and I connect again.
Thanks for your help and direction.

closed 2023-07-19 by bert269