HeyWillow/willow #221: Easy way to generate Long-lived access token for google assist

XRlabTech, 2023-07-03:
Hello

I have trying to use the following link to generate a long lived access token

https://developers.home-assistant.io/docs/auth_api/#:~:text=Long%2Dlived%20access%20tokens%20can,access%20token%20for%20current%20user.

Is they any other way to get one . I have been looking every where.

nikito, 2023-07-03:
Can be done in the UI:
[image]
You go into the side bar, Click on your user, and it will be all the way at the bottom. :)

XRlabTech, 2023-07-03:
Hi 

Do you mean through the google home app which specific dashboard? Please post link I an new to the whole google assist ecosystem.

nikito, 2023-07-03:
Not sure I understand? HA doesn't use google assistant for Longed Lived Tokens, it is all within Home Assistant itself.

XRlabTech, 2023-07-03:
Ok Thanks.

So Home Assistant Lovelace UI would be running on the host lunix similar to this link : ht1tp0s://www.home-assistant.io/installation/linux ?

nikito, 2023-07-03:
It's usually served by the HA server on port 8123, like http://homeassistant.local:8123/lovelace

stintel, 2025-02-23:
Question answered and no activity for > 1 year.

closed 2025-02-23 by stintel