HeyWillow/willow #222: Flashing issue Help : sudo PORT=/dev/ttyACM0 ./utils.sh flash

XRlabTech, 2023-07-03:
Hello

I was able to successfully erase flash the esp32-s3-box and when I run the commands below I get the following error
not sure what I am doing wrong. 

sudo PORT=/dev/ttyACM0 ./utils.sh flash
Using venv for esptool
esptool esp32s3 -p /dev/ttyACM0 -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 ui.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:d4:fe:a0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x001b0fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00210000 to 0x00680fff...
Flash will be erased from 0x00710000 to 0x0078ffff...
Compressed 22240 bytes to 13940...
Writing at 0x00000000... (100 %)Traceback (most recent call last):
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/__init__.py", line 1032, in _main
    main()
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/__init__.py", line 832, in main
    operation_func(esp, args)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/cmds.py", line 586, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/loader.py", line 131, in inner
    return func(*args, **kwargs)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/loader.py", line 1030, in flash_defl_block
    self.check_command(
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/loader.py", line 435, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/loader.py", line 404, in command
    p = self.read()
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/loader.py", line 337, in read
    return next(self._slip_reader)
StopIteration

A fatal error occurred: The chip stopped responding.

hamishcunningham, 2023-07-03:
Could just be an intermittent error. Have you tried restart / reconnect / new cable etc.?

XRlabTech, 2023-07-03:
Hi

I am using Virtual Box on lunix and try every thing I still keep getting.

sudo PORT=/dev/ttyACM0 ./utils.sh flash
[sudo] password for tokennodes: 
Using venv for esptool
esptool esp32s3 -p /dev/ttyACM0 -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 ui.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:d4:fe:a0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 2000000
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00010000 to 0x001b0fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00210000 to 0x00680fff...
Flash will be erased from 0x00710000 to 0x0078ffff...
Compressed 22240 bytes to 13940...
Writing at 0x00000000... (100 %)
A fatal error occurred: Packet content transfer stopped (received 2 bytes)

Second attempt usually results in:

udo PORT=/dev/ttyACM0 ./utils.sh flash
Using venv for esptool
esptool esp32s3 -p /dev/ttyACM0 -b 2000000 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x10000 willow.bin 0x8000 partition_table/partition-table.bin 0x210000 srmodels/srmodels.bin 0x710000 ui.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Connecting...
Traceback (most recent call last):
  File "/home/tokennodes/willow/venv/bin/esptool.py", line 34, in <module>
    esptool._main()
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/__init__.py", line 1032, in _main
    main()
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/__init__.py", line 674, in main
    esp = esp or get_default_connected_device(
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/__init__.py", line 921, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/loader.py", line 640, in connect
    last_error = self._connect_attempt(reset_strategy, mode)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/loader.py", line 525, in _connect_attempt
    reset_strategy()  # Reset the chip to bootloader (download mode)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/reset.py", line 100, in __call__
    self._setRTS(False)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/esptool/reset.py", line 40, in _setRTS
    self.port.setRTS(state)
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/serial/serialutil.py", line 600, in setRTS
    self.rts = value
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/serial/serialutil.py", line 463, in rts
    self._update_rts_state()
  File "/home/tokennodes/willow/venv/lib/python3.8/site-packages/serial/serialposix.py", line 708, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIC, TIOCM_RTS_str)
BrokenPipeError: [Errno 32] Broken pipe

I am got to try restarting the docker container.

nikito, 2023-07-03:
Kinda looks like the serial connection is being broken during the command. Could be some weirdness with virtualbox?

nikito, 2023-07-03:
Also before doing flash, did you do erase-flash? I would try that command again as well if a flash failed, as the device is likely in an inconsistent state.

XRlabTech, 2023-07-03:
Hi

The erase below worked:
 
sudo PORT=/dev/ttyACM0 ./utils.sh erase-flash

Its just when I try the flash.

Could pressing the mode button work?

nikito, 2023-07-03:
Device screen will remain black until it boots since it has no OS to run. Console should show info when flashing though. I haven't really run into anything like this flashing two devices several times each, but I also am running the willow container on a docker server in proxmox with the USB port directly fed to the VM. Not familiar enough with Virtualbox to know what could be going on there :/

XRlabTech, 2023-07-03:
This are the main errors usually appears 

A fatal error occurred: Packet content transfer stopped (received 1 bytes)

A fatal error occurred: The chip stopped responding.

nikito, 2023-07-03:
Yeah I have never seen either of those errors before when flashing. It really seems like the connection is just dropped. Maybe you can try the esp online flash tool? I think there's been some topics in here that discuss using that instead of a VM or WSL. Or you could try it natively in a linux machine if you have one available to try and rule out Virtualbox causing some sort of issue. The 1.0 release for Willow will have their own online flash tool as part of installation removing most of these issues, where this kind of flashing would only be for developers working on the code. 😃

XRlabTech, 2023-07-03:
Hi

I am using flash_download_tool_3.9.5 to flash the original image. I will try different cables and a native lunix machine. I will look out for version 1.0

stintel, 2023-08-17:
Closing as we now offer https://flash.heywillow.io/.

closed 2023-08-17 by stintel