HeyWillow/willow #227: LVGL rapid loops on scroll after many longer than display width text lines

kristiankielhofner, 2023-07-08:
When doing testing for commands that fail intent match in HA I've observed a rapid loop on the display for long lines that need to scroll. This tends to occur after around 70 test loops.

stintel, 2023-08-15:
As of commit a257314b9cd3243f0d1de64f5dc18d295343e1f7 in the feature/was branch, we should retest enabling text scrolling. We will do this after merging feature/was into main and tagging our first pre-release.

stintel, 2023-09-13:
Some things to try:
* reproduce with default LVGL font instead of Willow custom font
* bump esp_lvgl_port to 1.3.0
* bump lvlgl to 8.3.9
* IDF 5.0/5.1

kristiankielhofner, 2023-09-13:
I can confirm the issue still persists within 70 test loops with the default built-in font. I will try lvgl_port and LVGL bumps now.

stintel, 2023-10-02:
It appears having > 1 line of text with scrolling enabled reliably reproduces the issue for me.

E.g. this long and unsupported HA command:
How many lights are on in office?
With the following response:
Sorry, I couldn't understand that.

Both these lines are too long to fit on the display and that seems to trigger this bug every time.

closed 2025-04-24 by stintel