HeyWillow/willow #241: Input fields for the WAS URL and Wi-Fi credentials not appearing

nirnachmani, 2023-08-18:
I am following the new on-boarding instructions using WAS and Willow Web Flash but when I am using the flash, after connecting to willow, the Input fields for the WAS URL and Wi-Fi credentials do not appear. I did manage to do it once but then changed my WAS address (after installing WAS on a different computer) and trying to flash again, I am getting this issue. Here is a screenshot:

[image]

Any ideas?

stintel, 2023-08-18:
Does anything appear in your browser console?
Do you have any browser add-ons like ad blockers or similar?
Try running in incognito window?

nirnachmani, 2023-08-18:
Thanks. I do get an error in the console but I am not sure if they are relevant:
```
Uncaught (in promise) TypeError: jsonResponse is not iterable
    at getReleases (index.js:71:25)
    at async connectButton.onclick (index.js:153:14)
getReleases @ index.js:71
 ```

No add-ons installed. Issue persists in incognito mode.

nirnachmani, 2023-08-18:
I followed the error. When I try to open https://api.github.com/repos/toverainc/willow/releases in the browser I get an error (IP address redacted):

```
{"message":"API rate limit exceeded for xxx.xxx.xxx.xxx. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)","documentation_url":"https://docs.github.com/rest/overview/resources-in-the-rest-api#rate-limiting"}
```

stintel, 2023-08-18:
Good catch! We're using a Cloudflare Worker in WAS to avoid hitting the Github rate-limit, but we forgot to update Willow Web Flasher to use this.