HeyWillow/willow #243: Use one port for the server and its config

skorokithakis, 2023-08-20:
It would be easier if we could deploy the server endpoint (the websocket) and the config endpoint under one URL, as now we need to deploy two hostnames, two TLS certs, etc.

The server could run under /ws for the websocket path (as it does now) and /config for the config server, instead of two different ports.

kristiankielhofner, 2023-08-20:
I'm a little confused by this - are you referring to the Willow Application Server endpoints?

If so, the UI currently uses Streamlit because @stintel and I don't have any frontend experience. The Streamlit UI should generally be considered a minimum PoC interface. Our goal is to work with the community to reimplement the Streamlit UI with a modern TS/JS framework that interacts with the existing WAS API.

With this approach we can mount the output HTML+JS+CSS path using FastAPI and eliminate Streamlit and the additional listening socket it uses.

skorokithakis, 2023-08-20:
@kristiankielhofner I am, yes. Does Streamlit prevent you from listening to the websocket URL in the same server?

Basically, my ideal workflow would be that the Streamlit UI is served under `http://localhost:8501`, and the websocket under `ws://localhost:8501/ws`. Is this possible right now?

Also, I think that a modern JS framework for a simple configuration interface might be a bit too heavy, especially for a resource-strapped OSS project. Do you absolutely need to be changing the settings over the WAS API? If not, you could write a very simple UI with some ready-made template with Flask and HTMX in a few hours and call it a day.

kristiankielhofner, 2023-08-20:
As best as we can tell, yes. We also didn't spend that much time investigating it because Streamlit has a variety of other issues that make it less than ideal for the WAS use case.

Yes, the API is necessary because another goal is to integrate WAS with Home Assistant and any other platforms to provide functionality and tighter integration with those platforms similar to esphome (in the case of Home Assistant).

skorokithakis, 2023-08-20:
> As best as we can tell, yes. We also didn't spend that much time investigating it because Streamlit has a variety of other issues that make it less than ideal for the WAS use case.

That makes sense, thanks.

> Yes, the API is necessary because another goal is to integrate WAS with Home Assistant and any other platforms to provide functionality and tighter integration with those platforms similar to esphome (in the case of Home Assistant).

*Having* the API is necessary, but do you need to be changing the settings over it (rather than writing to the database directly, for example)? It would be much simpler to write one or two views with plain HTML than have to maintain an entirely separate TS/JS codebase, but I guess it also depends on how much fancy you want the config UI to get.

kristiankielhofner, 2023-08-21:
Generally we build for where we're going, not where we are.

WAS is very much in its infancy. Long term we plan to implement a dedicated protocol utilizing websockets for all Willow functions to enable all kinds of interesting functionality (for things like your ChatGPT use case) where the flow looks more like:

Willow websocket to WAS -> speech to any STT platform -> interception of transcription -> any command platform/API/etc -> TTS of your choice -> output to Willow.

This has the benefit of not only being significantly more flexible but also reducing the complexity of the various command implementations (HA, OpenHAB, REST, etc) running on the Willow device.

Hopefully this helps to better understand our approach - the UI for this functionality will be significantly more complex and the API backend is an obvious requirement.

skorokithakis, 2023-08-21:
That makes sense, thanks for the info!

closed 2023-08-21 by skorokithakis