HeyWillow/willow #244: Willow won't start

skorokithakis, 2023-08-20:
I flashed Willow from the web flasher, but I get this error when starting:

```
I (00:00:20.841) WILLOW/MAIN: Starting up! Please wait...
I (00:00:20.926) WILLOW/MAIN: SPIFFS mounted
I (00:00:20.926) WILLOW/CONFIG: opening /spiffs/user/config/willow.json
E (00:00:20.988) WILLOW/CONFIG: failed to open /spiffs/user/config/willow.json
W (00:00:25.068) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (00:00:26.568) WILLOW/WAS: initializing WebSocket client (wss://willowserver.home.stavros.io/ws)
I (00:00:26.569) WILLOW/WAS: initializing WebSocket client (wss://willowserver.home.stavros.io/ws)

***ERROR*** A stack overflow in task websocket_task has been detected.

Backtrace: 0x4037a1be:0x3fcc4b50 0x40383fed:0x3fcc4b70 0x40387a1a:0x3fcc4b90 0x40385d68:0x3fcc4c00 0x403840a4:0x3fcc4c20 0x4038409a:0x4148e269 |<-CORRUPTED

ELF file SHA256: 5622c060075d1bb1

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x42146a92
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x170c
load:0x403c9700,len:0xc14
load:0x403cc700,len:0x33b4
entry 0x403c9968
```

It looks like the willow.json is missing, but why? Could it be because there's a space in my wifi SSID?

skorokithakis, 2023-08-20:
I have the same issue after spending a few hours trying to build the image from scratch. It can never find the JSON config file.

nikito, 2023-08-20:
I think the willow.json may be a red herring here; if I recall correctly this error is expected on first flash since the config hasn't been pulled from WAS yet. One thing I noticed is that the ssid for Wi-Fi is case sensitive (I was getting weird connecting until I made sure the ssid entered exactly matched what my ap's were advertising). Not sure if spaces in the SSID matter either, I personally don't spaces in mine.

skorokithakis, 2023-08-20:
Aw, I've spent so much time on this, and I think I just managed to pack the `willow.json` into an SPIFFS image that I am now flashing. Too bad about the error message, but then I'm lost on why it doesn't work...

On re-reading the error message, you're right:

```
A stack overflow in task websocket_task has been detected.
```

Ugh :( Thank you.

skorokithakis, 2023-08-20:
After changing the application server a few times, it still doesn't work. In the end, I deployed the server directly to a port, and had willow talk to it directly, instead of behind a reverse proxy, but no dice:

```
I (436) esp_image: segment 4: paddr=001dc234 vaddr=4037beW (00:00:04.364) PERIPH_WIFI: WiFi Event cb, Unhandle event_base:WIFI_EVENT, event_id:4
I (00:00:05.865) WILLOW/WAS: initializing WebSocket client (http://192.168.10.4:32072)
E (00:00:05.866) WEBSOCKET_CLIENT: There are no transports valid, stop websocket client
I (00:00:05.871) WILLOW/WAS: initializing WebSocket client (http://192.168.10.4:32072)
E (00:00:05.880) WEBSOCKET_CLIENT: There are no transports valid, stop websocket client
E (00:00:05.880) WEBSOCKET_CLIENT: Websocket client is not connected
E (00:00:05.896) WILLOW/WAS: failed to send WAS get_config message
```

Willow is on the wifi network, and I can ping it, and that host is up and responding.

skorokithakis, 2023-08-20:
Ok, it works after appending `/ws` to the URL. It didn't work when I was using any kind of reverse proxy, but running the bare Docker server seems to work.

This could all have been a bit better-documented, but at least it works now.

closed 2023-08-20 by skorokithakis