HeyWillow/willow #246: Allow the REST server to return TTS text

skorokithakis, 2023-08-20:
It would be useful if the REST server could return a `speech` key in the JSON object, to be spoken on the device, along with the `text` key it returns now.

skorokithakis, 2023-08-20:
To enable this amazing application:

https://twitter.com/Stavros/status/1693204822042739124/mediaViewer?currentTweet=1693204822042739124&currentTweetUser=Stavros

skorokithakis, 2023-08-20:
Apparently #225 is related.

kristiankielhofner, 2023-09-25:
We support this in our rc2 release candidate. Can you upgrade and confirm?

skorokithakis, 2023-09-25:
@kristiankielhofner yep, definitely! Do you have some documentation on what the API should return? Thanks!

kristiankielhofner, 2023-09-25:
It currently expects an HTTP response in plain text. Whatever text your REST endpoint returns is the text printed on the display and passed for TTS.

In the near future we are moving all of the command endpoint integrations to WAS with a single WAS endpoint in Willow. When this happens we will have significantly better parsing of responses from the various endpoints and we'll ensure it's backward compatible with the very limited REST parsing we have today. This will also enable users to define their own command endpoints in WAS using Python which is significantly easier, more capable, less error prone, and flexible than requiring individual endpoint support in Willow itself. We could (for example) have native ChatGPT, etc command endpoints.

WAS is called the Willow Application server because our long-term plan is to enable users to share command endpoint and other integrations they have developed with the community through what would essentially be a Willow App repository.

skorokithakis, 2023-09-25:
Ah, that sounds good, thank you!

skorokithakis, 2023-09-27:
@kristiankielhofner I've just tried this and it works perfectly, thanks! The "thinking" is a nice touch as well, much better UX.

kristiankielhofner, 2023-09-28:
Great! Thanks, we have some other UX improvements coming, stay tuned.