HeyWillow/willow #247: Show more information while recognizing

skorokithakis, 2023-08-20:
Currently, the Willow only shows "say command" until the response is back. It would be better to show something like "processing...", as my setup might take a few seconds to respond, and showing "say command" during a time when the device isn't listening is confusing UX.

kristiankielhofner, 2023-08-24:
How do you feel about displaying "Processing..." on the same line as "Say command..." after speech ends and when the command is sent to the command endpoint?

skorokithakis, 2023-08-24:
You mean instead of "say command..."? Yes, that was what I meant, though on second thought "processing" might be a bit too jargony. Depending on your intended audience, maybe "thinking" is more understandable?

kristiankielhofner, 2023-08-24:
Thanks for understanding, glad to hear you won't be asking for your money back ;).

I'm actually embarrassed we didn't think of this before - you're absolutely correct in that it's good to get some definitive feedback on when voice has ended, something is happening, and the user just needs to wait. Of course in my testing with Home Assistant and a local RTX 3090 "Thinking..." basically flashes on the display for a couple hundred ms but it's still good to know.

Your ChatGPT + Willow Twitter video is really cool but sitting there at "Say command..." is pretty cringe to me and this is an easy change that makes a huge difference in user experience.

If I push this change to a branch do you have a build/dev environment setup to test? I've already put it through our automated testing for 2,000 successive runs but it would be good to test with your configuration as well.

skorokithakis, 2023-08-24:
I can test it, yep! I'll need to make some changes to the code to make the Box speak the result, and flashing my changes has been a bit finicky, but I should be able to test, yes.

skorokithakis, 2023-09-04:
@kristiankielhofner do you have a branch for this, perhaps?

closed 2023-09-05 by kristiankielhofner

skorokithakis, 2023-09-05:
I'll try this, thanks!

skorokithakis, 2023-09-05:
Oof, amazing :kissing: :ok_hand:.

PR #255 (2023-09-05): Print "Thinking..." on the display after speech while command is sent
Continuing to display "Say Command..." after speech has ended and the command has been sent to the command endpoint is poor and confusing UI. Print "Thinking..." after speech has ended while we're waiting for response.

Fixes #247