HeyWillow/willow #254: Voice recognition often stops prematurely

skorokithakis, 2023-09-03:
Willow will often stop recognizing mid-sentence, especially if I pause slightly to find my next word. It's very trigger-happy with stopping, to the extent where I think about what I want to say in its entirety before I talk to Willow.

nikito, 2023-09-04:
Have you tried increasing the VAD timeout? By default it is 300ms, meaning if you stop talking for 300ms it assumes you are done with your command. You can raise this higher which would give you more time before it stops listening. 🙂

skorokithakis, 2023-09-04:
Ah, I didn't realize this was an option, thank you. Perhaps it can be documented better, in a troubleshooting FAQ or something similar? Or is it already and I just made a fool of myself? 😛

closed 2023-09-04 by skorokithakis